bblanchon/ArduinoJson #1494: Deserializing from EEPROM I get the first char only from char[]

dentex, 2021-02-04:
Hello and thanks for the awesome tools you designed.

I'm playing with an Arduino Pro Micro and into the EEPROM I have loaded:
```
{
	"name": "Apps",
	"keypad": {
		"btn0": {
			"key": "KEY_Z",
			"ctrl": true,
			"shift": false,
			"label": "     UNDO      "
		},
		"btn1": {
			"key": "KEY_Z",
			"ctrl": true,
			"shift": true,
			"label": "     REDO      "
		}
	}
}
```

I'm using the subsequent code for deserialization:
```
  EepromStream es(0, 1024);
  StaticJsonDocument<1024> doc;
  deserializeJson(doc, es);

  const char *name = doc["name"]; // "Apps"

  JsonObject keypad_btn0 = doc["keypad"]["btn0"];
  const char *keypad_btn0_key = keypad_btn0["key"];     // "KEY_Z"
  bool keypad_btn0_ctrl = keypad_btn0["ctrl"];          // true
  bool keypad_btn0_shift = keypad_btn0["shift"];        // false
  const char *keypad_btn0_label = keypad_btn0["label"]; // "     UNDO      "

  JsonObject keypad_btn1 = doc["keypad"]["btn1"];
  const char *keypad_btn1_key = keypad_btn1["key"];     // "KEY_Z"
  bool keypad_btn1_ctrl = keypad_btn1["ctrl"];          // true
  bool keypad_btn1_shift = keypad_btn1["shift"];        // true
  const char *keypad_btn1_label = keypad_btn1["label"]; // "     REDO      "

  // Print values.
  Serial.println(name);

  Serial.println(*keypad_btn0_key);
  Serial.println(keypad_btn0_ctrl);
  Serial.println(keypad_btn0_shift);
  Serial.println(*keypad_btn0_label);

  Serial.println(*keypad_btn1_key);
  Serial.println(keypad_btn1_ctrl);
  Serial.println(keypad_btn1_shift);
  Serial.println(*keypad_btn1_label);
}
```

The problem is that in the Serial console I get the output:
```
Apps      <--- CORRECT
K         <--- 1st char only from "KEY_Z"
1
0
          <--- 1st char only from "     UNDO      "
K         <--- 1st char only from "KEY_Z"
1
1
          <--- 1st char only from "     REDO      "
```

What am I missing that makes me retrieve only the 1st chars from those values?

Thanks.

bblanchon, 2021-02-05:
Hi @dentex,

Thank you very much for reporting this issue.
I managed to reproduce it on my Arduino Leonardo.
I'll get back to you soon.

Best regards,
Benoit

bblanchon, 2021-02-05:
OK, I just found the bug: it's in your program.

To print a string, you must not dereference the char pointer like so:

```c++
// BAD: prints the first char
Serial.println(*keypad_btn0_key);
```

Instead, you must pass the char pointer to `println()`:

```c++
// GOOD: prints the string
Serial.println(keypad_btn0_key);
```

dentex, 2021-02-05:
OK!
...I knew it was something n00b from my side...
 Thank you very much!

bblanchon, 2021-02-05:
You're welcome, @dentex.
Thank your for using ArduinoJson.
Good luck with your project!

closed 2021-02-05 by bblanchon