bblanchon/ArduinoJson #1495: ArduinoJson6 Assistant for MessagePack

gnalbandian, 2021-02-05:
Hi. 
When using MessagePack, which is the best way to calculate DynamicJsonDocument size? 
Should we use the json assistant?

I believe there could an assistant for MessagePack since the resulting serialized data is smaller than plain json.

bblanchon, 2021-02-05:
Hi @gnalbandian,

MessagePack doesn't affect the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/)'s size, so all the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/)'s results are still relevant.
Only the last step, which shows a sample program, could be changed to call [`serializeMsgPack()`](https://arduinojson.org/v6/api/msgpack/serializemsgpack/) instead of [`serializeJson()`](https://arduinojson.org/v6/api/json/serializejson/), but I don't think this is worth the effort.
I don't see anything else to do.

Best regards,
Benoit

gnalbandian, 2021-02-06:
Thanks @bblanchon. I was wrong then. You're right. It isn't worth while.
Regards!
Gonzalo

closed 2021-02-06 by gnalbandian