bblanchon/ArduinoJson #1496: Nano and ESP28J60

Pyroa, 2021-02-05:
Hi,

I use the arduinojson library on Arduino IoT 33, ESP8266 And Nano.

My issue is happening on the nano with a ENC28J60 ethernet shield.

By the way, cause of the small size of the memory i spent lot of time to find a working program. But ! something went wrong.

Typically, my program received a json string from my computer through the LAN. I deserialized it. Sometimes it's ok, sometimes i've got IncompletInput error ...
But all the time, if the first string is deserialized and the response sent to correctly to my computer, if i resend a second time a new request ... the nano reset or totally crash. Need to be reset

I don't understand. i suppose that the memory is full but something is missing to me to understand why.

Can you help ?

Regards,

Pyroa

Ps : Sourcecode in attached file
[main.txt](https://github.com/bblanchon/ArduinoJson/files/5934165/main.txt)

bblanchon, 2021-02-06:
Hi Pyroa,

[`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput) occurs because your program reads too few bytes. `client.available()` doesn't contain the size of the complete message, only what has been received so far.
Instead of reading the buffer from your program, let [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) pull from the stream.
It's not only simpler, but it will also reduce memory usage and eliminate the need for heap memory.
See [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/) for an example.
Read the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/) for a complete explanation.

A crash or a reset is often due to a dangling pointer. For example, I spotted the following mistakes in your program:

```c++
const char *Message = Request["Name"];   // null if "Name" is missing
char *Append = String(" commmand is unkown").c_str();  // pointer to a freed buffer + ignore constness

char Output[(sizeof(Message) + sizeof(Append))];  // wrong size calculation

strcpy(Output,Message);  // buffer overflow + null pointer dereferencing
````

**That's six bugs in four lines of code !!!**
Please don't ignore the compiler warnings.
Please read [the documentation](https://arduinojson.org/v6/doc/) and [the examples](https://arduinojson.org/v6/example/).
If you need a short course on C++, please read chapter 2 of [Mastering ArduinoJson](https://arduinojson.org/book/)
See also [Why does my device crash or reboot?](https://arduinojson.org/v6/issues/crash-or-reboot/)

I don't think memory is full since your JsonDocuments are very small.
You don't need to call [garbageCollect()](https://arduinojson.org/v6/api/jsondocument/garbagecollect/) at the end of the scope; the JsonDocument is discarded anyway.
Only use this function when you keep the JsonDocument in memory for a long period, which you shouldn't do anyway.

Best regards,
Benoit

Pyroa, 2021-02-06:
Hi, First tanks for your answer 

> Hi Pyroa,
> 
> [`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/#incompleteinput) occurs because your program reads too few bytes. `client.available()` doesn't contain the size of the complete message, only what has been received so far.

Yes i know that, i'm just trying  in first to correctly deserialized.

> Instead of reading the buffer from your program, let [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) pull from the stream.
> It's not only simpler, but it will also reduce [...]

Yes it's was my first usage of the library. But it's not working.

DeserializationError error = deserializeJson(ArduinoResquest, client);

Working on ESP8266,and IOT 33. But not on the nano.

> 
> A crash or a reset is often due to a dangling pointer. For example, I spotted the following mistakes in your program:
> > ```c++

Omg, i'm stupid ... too tired certainly ... :'(
Focused on an issue, i forgot the rest ...

> const char *Message = Request["Name"];   // null if "Name" is missing
> char *Append = String(" commmand is unkown").c_str();  // pointer to a freed buffer + ignore constness
>
> char [...]
yes indeed. Sorry it's was code not reviewed and directly write without thinking.
next time i remove it from the exemple to avoid to mess your time , and mess mine !!!!

> I don't think memory is full since your JsonDocuments are very small.
> You don't need to call [garbageCollect()](https://arduinojson.org/v6/api/jsondocument/garbagecollect/) at the end of the scope; [...]

I've re-used DeserializationError error = deserializeJson(ArduinoResquest, client);
Without a dynamicJson object unlike in my IoT33 program, and it's working.

But crash happend after severals deserialization (shit code was removed)

i'm reading your tutorials and i will write o good code at head rested.

Thanks for your time.

closed 2021-02-20 by bblanchon