bblanchon/ArduinoJson #1497: Incompatible versions need new numbers!  This is a mess!!

MikeyMoMo, 2021-02-05:
You have made many updates to the JSON routines and that's great but you did not change the name of the .cpp and .h files.  This means that the revisions 0, 5 and 6 (and maybe others) ALL have the same names for the files.  The folders have different names but that's not how the Arduino IDE and C work.  They go by the .h name.  When there are 3 or more libraries (since they have different names, they can coexist in the libraries folder) and they are all named <ArduinoJson.h> and different programs written at different times want different versions, it is a total train wreck.  One has to swap JSON routines folder in and out and hope to get the version the program wants and then do it again for a program wanting a different version.

PLEASE (and it is not too late) rename the .h and .cpp files to have a version number on them.  Then, ArduinoJson5.h and ArduinoJson6.h would not collide and cause wasted time swapping out versions.  Rewriting programs is an option, of course, but for something downloaded that one just wants to compile and use, it is not something that anyone really wants to do.

All of this hassle is easily fixed.  Then when and if you write another major, incompatible version, please name it ArduinoJson7.h and .cpp, PLEASE!

bblanchon, 2021-02-06:
Hi @MikeyMoMo,

Thank you for this kind and tactful request.

I already thought about that. The problem is that just renaming the header won't allow you to install several library versions, at least not with the Arduino Library Manager.
To allow this, I should rename the library itself. Although it might be possible, I don't really know all the implications.
Besides, other library managers, like PlatformIO's and Particles's already fixed the problem by allowing the user to specify the version per project (like any good package manager).

Don't worry; I still have a solution if you want to stick to the Arduino IDE: commit the library's single header version with your source files. Here are all the versions available:

* [`ArduinoJson-v5.8.4.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.8.4/ArduinoJson-v5.8.4.h)
* [`ArduinoJson-v5.9.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.9.0/ArduinoJson-v5.9.0.h)
* [`ArduinoJson-v5.10.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.10.0/ArduinoJson-v5.10.0.h)
* [`ArduinoJson-v5.10.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.10.1/ArduinoJson-v5.10.1.h)
* [`ArduinoJson-v5.11.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.11.0/ArduinoJson-v5.11.0.h)
* [`ArduinoJson-v5.11.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.11.1/ArduinoJson-v5.11.1.h)
* [`ArduinoJson-v5.11.2.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.11.2/ArduinoJson-v5.11.2.h)
* [`ArduinoJson-v5.12.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.12.0/ArduinoJson-v5.12.0.h)
* [`ArduinoJson-v5.13.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.0/ArduinoJson-v5.13.0.h)
* [`ArduinoJson-v5.13.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.1/ArduinoJson-v5.13.1.h)
* [`ArduinoJson-v5.13.2.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.2/ArduinoJson-v5.13.2.h)
* [`ArduinoJson-v5.13.3.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.3/ArduinoJson-v5.13.3.h)
* [`ArduinoJson-v5.13.4.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.4/ArduinoJson-v5.13.4.h)
* [`ArduinoJson-v5.13.5.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.5/ArduinoJson-v5.13.5.h)
* [`ArduinoJson-v6.0.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.0.0-beta/ArduinoJson-v6.0.0-beta.h)
* [`ArduinoJson-v6.0.1-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.0.1-beta/ArduinoJson-v6.0.1-beta.h)
* [`ArduinoJson-v6.1.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.1.0-beta/ArduinoJson-v6.1.0-beta.h)
* [`ArduinoJson-v6.2.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.2.0-beta/ArduinoJson-v6.2.0-beta.h)
* [`ArduinoJson-v6.2.1-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.2.1-beta/ArduinoJson-v6.2.1-beta.h)
* [`ArduinoJson-v6.2.2-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.2.2-beta/ArduinoJson-v6.2.2-beta.h)
* [`ArduinoJson-v6.2.3-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.2.3-beta/ArduinoJson-v6.2.3-beta.h)
* [`ArduinoJson-v6.3.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.3.0-beta/ArduinoJson-v6.3.0-beta.h)
* [`ArduinoJson-v6.4.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.4.0-beta/ArduinoJson-v6.4.0-beta.h)
* [`ArduinoJson-v6.5.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.5.0-beta/ArduinoJson-v6.5.0-beta.h)
* [`ArduinoJson-v6.6.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.6.0-beta/ArduinoJson-v6.6.0-beta.h)
* [`ArduinoJson-v6.7.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.7.0-beta/ArduinoJson-v6.7.0-beta.h)
* [`ArduinoJson-v6.8.0-beta.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.8.0-beta/ArduinoJson-v6.8.0-beta.h)
* [`ArduinoJson-v6.9.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.9.0/ArduinoJson-v6.9.0.h)
* [`ArduinoJson-v6.9.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.9.1/ArduinoJson-v6.9.1.h)
* [`ArduinoJson-v6.10.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.10.0/ArduinoJson-v6.10.0.h)
* [`ArduinoJson-v6.10.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.10.1/ArduinoJson-v6.10.1.h)
* [`ArduinoJson-v6.11.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.0/ArduinoJson-v6.11.0.h)
* [`ArduinoJson-v6.11.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.1/ArduinoJson-v6.11.1.h)
* [`ArduinoJson-v6.11.2.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.2/ArduinoJson-v6.11.2.h)
* [`ArduinoJson-v6.11.3.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.3/ArduinoJson-v6.11.3.h)
* [`ArduinoJson-v6.11.4.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.4/ArduinoJson-v6.11.4.h)
* [`ArduinoJson-v6.11.5.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.11.5/ArduinoJson-v6.11.5.h)
* [`ArduinoJson-v6.12.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.12.0/ArduinoJson-v6.12.0.h)
* [`ArduinoJson-v6.13.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.13.0/ArduinoJson-v6.13.0.h)
[... cut ...]
* [`ArduinoJson-v6.17.0.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.17.0/ArduinoJson-v6.17.0.h)
* [`ArduinoJson-v6.17.1.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.17.1/ArduinoJson-v6.17.1.h)
* [`ArduinoJson-v6.17.2.h`](https://github.com/bblanchon/ArduinoJson/releases/download/v6.17.2/ArduinoJson-v6.17.2.h)

In the future, you can always find the latest version at the bottom of [this page](https://github.com/bblanchon/ArduinoJson/releases/latest) or in [this page](https://arduinojson.org/v6/doc/installation/).

Best regards,
Benoit

bblanchon, 2021-02-19:
I added the following pages :

* https://arduinojson.org/v5/revisions/
* https://arduinojson.org/v6/revisions/

They contain links to the older versions of the library.
Hopefully, it will help users find this information more easily.

closed 2021-02-19 by bblanchon