bblanchon/ArduinoJson #1498: Problem with aggregate initialization of std::string

mattiasm83, 2021-02-06:
There seems to be a problem with aggregate initialization of std::string.

The variables b and c in the code below uses this:
```
std::string a = mqttConf[F("server")];
std::string b = {mqttConf[F("server")]};
std::string c{mqttConf[F("server")]};

Serial.println("----");
Serial.println(a.c_str());
Serial.println(b.c_str());
Serial.println(c.c_str());
Serial.println("----");
```
The output of the following code shows this:
```
----
192.168.1.1

----
```
Only the first initialization works.

It this a known limitation, or is it a bug?

bblanchon, 2021-02-07:
Hi @mattiasm83,

Indeed, the compiler picks the wrong overload.
The call is equivalent to:

```c++
std::string c{mqttConf[F("server")].as<char>()};
```

Unfortunately, I have no way to make this statement work as you expect.
This is the price to pay for the convenience of implicit conversion operators.

To address this issue, I explicitly forbid `JsonVariant::as<char>()` to that this line causes a compilation error.
I know this is quite deceptive, but it's still better than a program that pretends to work and doesn't.
I also updated [the documentation](https://arduinojson.org/v6/api/config/enable_std_string/) to show when the error occurs and how to work around it.

Since this is a breaking change, I'll wait for 6.18 to publish it.

Best regards,
Benoit

mattiasm83, 2021-02-07:
Thanks for the quick reply!
The solution you made seems good. I agree that I rather have an error show up in compilation time, than in runtime.

Keep up the good work with the lib :+1:

closed 2021-02-18 by bblanchon

bblanchon, 2021-05-05:
Fix published in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

commit 5234c812 (2021-02-18): Removed `JsonVariant::as<char>()` (fixes #1498)