bblanchon/ArduinoJson #1499: Cannot parse String Array within a nested object

carthoris, 2021-02-07:
I have a structure as follows within a larger JSON:

"shapes": [
		{
			"type": "polyline",
			"name": "polyline1",
			"size": 1,
			"loopback": true,
			"edges": ["segment1"]
		}
	]
I have successfully parsed out everything except for the "edges" (array of strings).
I have copied the code generated by the arduinojson assistant and it still does not work:

JsonObject shapes_0 = doc["shapes"][0];
const char* shapes_0_type = shapes_0["type"]; // "polyline"
const char* shapes_0_name = shapes_0["name"]; // "polyline1"
int shapes_0_size = shapes_0["size"]; // 1
bool shapes_0_loopback = shapes_0["loopback"]; // true

const char* shapes_0_edges_0 = shapes_0["edges"][0]; // "segment1"
Log.warning("edge name: %s" CR, shapes_0_edges_0);

This returns a blank value.

I also tried
JsonArray edgesList = doc["shapes"][0]["edges"].as<JsonArray>();
and edgestList size is 0

also
JsonArray shapeList = doc["shapes"].as<JsonArray>();
JsonArray edgesList = shapeList[shapeCount]["edges"].as<JsonArray>();
and edgestList size is 0

Kind of at a loss how to get to the string array.

Thanks in advance.

bblanchon, 2021-02-08:
Hi,

I tried

```c++
JsonArray edgesList = doc["shapes"][0]["edges"];
```

and I can confim that `edgesList.size()` returns 1.

Are you sure that your [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) is large enough?
What is the status returned by [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/)?
Did you configure the ArduinoJson Assistant correctly, on the first page in particular?

Best regards,
Benoit

carthoris, 2021-02-08:
Thanks for the response. Will hook it up today and check all your questions.

carthoris, 2021-02-08:
Still not working for me. I did all of the above. Here is appropriate code:

 DynamicJsonDocument doc(2048);
 DeserializationError error = deserializeJson(doc, json);
 Log.warning("deserializeJson error %s" CR, error.c_str());

JsonArray edgesList1 = doc["shapes"][0]["edges"];
Log.warning("edge size: %d" CR, edgesList1.size());

JsonObject shapes_0 = doc["shapes"][0];
const char* shapes_0_type = shapes_0["type"]; // "polyline"
const char* shapes_0_name = shapes_0["name"]; // "polyline1"
int shapes_0_size = shapes_0["size"]; // 1
bool shapes_0_loopback = shapes_0["loopback"]; // true

const char* shapes_0_edges_0 = shapes_0["edges"][0];
Log.warning("edge name: %s" CR, shapes_0_edges_0);
		
and result:
W: deserializeJson error Ok
W: edge size: 0
W: edge name: 

and here is the entire JSON

{
	"colors": [
		{
			"name": "color1",
			"red": 0,
			"green": 0,
			"blue": 0
		},
		{
			"name": "color2",
			"red": 0,
			"green": 0,
			"blue": 127
		}
	],
	"speeds": [
		{
			"name": "speed1",
			"dist": 30,
			"ms": 1000,
			"units": "PIXELS"
		}
	],
	"paints": [
		{
			"name": "background",
			"color": "color1"
		}
	],
	"strips": [
		{
			"name": "strip1",
			"ledCount": 300
		}
	],
	"segments": [
		{
			"name": "segment1",
			"length": 1,
			"size": 300
		}
	],
	"shapes": [
		{
			"type": "polyline",
			"name": "polyline1",
			"size": 1,
			"loopback": true,
			"edges": ["segment1"]
		}
	],
	"actors": [
		{
			"name": "tracer1",
			"type": "tracer",
			"color": "color2",
			"speed": "speed1",
			"target": "polyline1"
		}
	]
}

bblanchon, 2021-02-09:
I just checked again with the new input and it works fine.
See for yourself here: https://wandbox.org/permlink/RC6YwpDigcqt77Yy
To move forward, I'd need an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) and some information about your hardware...

carthoris, 2021-02-09:
First I want to say thank you for your helpful responses.

Using MCVE style I was able to track down what works and what doesn't. 
Works:                const char* json = R"({.....})";
Doesn't work:    String json = "{.....}";

Is this a bug or am I using String incorrectly for this purpose?

bblanchon, 2021-02-10:
Using a string class without raw string literal changes nothing.
See for yourself here: https://wandbox.org/permlink/h5WR5tehocVCUNyT

carthoris, 2021-02-10:
Will keep trying

closed 2021-03-12 by bblanchon