bblanchon/ArduinoJson #1500: How do I pass a json doc built on mega to esp8266 so I can send out to HTTP client

edro1961, 2021-02-12:
Sweet mother of Mary....I've spent so much time trying to get this to work.. I'm losing my mind...lol So here I am...

Ok here's what I got..  I have a Mega 2560 with 10 sensors on it 
I have the esp8266 running UDPClient, ESP8266WiFi as client, and HTTPClient.
I've programmed the ESP to go get my NTP time from my local server. ( I set the mega to request it )
I then take the returned time and send it back to the Mega. ( This will become my first Json element).
I read the Time and the sensor data and then  using DynamicJsonDocument I create the doc. Everything is good so far.
Now I tell the ESP that I'm sending the posted Json doc so the ESP can apply the Rest API headers prior to the actual json data.
I'm sending the Json doc via serializejson( doc, Serial2) which is the serial that's tied to the ESP.... It gets sent but its not formatted correctly anymore for me to forward on to the HTTP Client.... Whew this got complex in a hurry.. Can you help me?

I can upload my ugly code ..lol if it would help

Thanks,

--Ed--

bblanchon, 2021-02-12:
Hi Ed,

It looks like the only relevant part here is:

> I'm sending the Json doc via serializejson( doc, Serial2)... It gets sent but its not formatted correctly

I'm assuming you meant that the JSON document is correctly sent but incorrectly received.
This sounds like a mismatch in the baud rates but could also come from the voltage: Mega is 5V, whereas ESP8266 is 3.3V.

See also:

* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit

edro1961, 2021-02-26:
Hi Benoit,

I've now overhauled program but now my snag point is the the deserializeJson won't read more that 64 bytes off the serial
Is this how it supposed to function or am I missing something?

Thanks in advance,

--Ed--

bblanchon, 2021-02-27:
Hi Ed,

It's difficult to tell what wrong with your code without looking at it.
Would you mind sharing a (small) piece of it?

Best regards,
Benoit

closed 2021-03-12 by bblanchon