bblanchon/ArduinoJson #1503: ArduinoJson - deserializeJson - wrong decimal values - esp8266

ieferrari, 2021-02-15:
Hello, I am using a NODE MCU 1.0 (ESP-12E Module) with ArduinoJson 6.17.3
I m trying to get a double value from a json payload but  the decimal numbers are wrong.

// CODE:
#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);

  char  line[]="{\"val1\":199999.11,\"val2\":599999.11,\"val3\":999999.11}";
  
  StaticJsonDocument<200> doc;
  DeserializationError error = deserializeJson(doc, line);

  double val1 = doc["val1"];
  double val2 = doc["val2"];
  double val3 = doc["val3"];  

  Serial.println(val1);
  Serial.println(val2);
  Serial.println(val3);
}

void loop() {
}

// Serial Output
199999.11
599999.13
999999.00

I keep getting no decimal values for big numbers.. and I cant figure out where is the problem
Thanks in advance

bblanchon, 2021-02-16:
Hi,

This is due to the lack of precision of 32-bits floating-point values.
You need to switch to double precision by setting [ARDUINOJSON_USE_DOUBLE](http://127.0.0.1:4000/v6/api/config/use_double/) to `1`.

I modified the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to warn about this problem.

Best regards,
Benoit

ieferrari, 2021-02-17:
I added:

#define ARDUINOJSON_USE_DOUBLE 1

(before  #include <ArduinoJson.h> ) 
and now it's working fine

Thanks for your help!

bblanchon, 2021-02-17:
You're welcome; thank you for using ArduinoJson.
Don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show you support 😉 
Good luck with your project!

closed 2021-02-17 by bblanchon