bblanchon/ArduinoJson #1505: Need help

sorokolokarim, 2021-02-19:
I want to deserialize and get value by this way but it doesn't work.

const char data[] = R"=({"x1":"1","x2":"2"})=";

int8_t value(const char* data, JsonDocument& jsonDoc, JsonVariant key)
{
	deserializeJson(jsonDoc, data);

	return key.as<int8_t>();
}

int8_t x  = value(data, jsonDoc, jsonDoc["x2"].to<JsonVariant>());
Serial.print("x : "); Serial.println(x);

// Output x : 1

Question 1 -> Why Output (x) = 1 and not 2
Question 2 -> How to do to have Output (x) = 2

Very very good library, thank you !
Sorry for my english, i speak french

bblanchon, 2021-02-20:
Hi @sorokolokarim,

It looks like you want something like this:

```c++
template<typename T>
T value(const char* json, const char* key)
{
   DynamicJsonDocument doc(4096);
   deserializeJson(doc, json);
   return doc[key].as<T>();
}

const char data[] = R"=({"x1":"1","x2":"2"})=";
int8_t x = value<int8_t>(data, "x2");
```

This function works with all types except `const char*`, but you can use `String` instead.

Best regards,
Benoit

bblanchon, 2021-02-20:
Online demo: https://wandbox.org/permlink/0Un6VcidHMhDBRx6

sorokolokarim, 2021-02-20:
Thank for your fast response. But effectively the document is biggest, this is just an exemple.  
So how to get the value of  x2 in this case and by the same way ?

const char data[] = R"=( {
    "a": {
        "b": {
            "x2": 2
        }

    }
})=";

Thank you

bblanchon, 2021-02-21:
Hi @sorokolokarim,

Doing so require a recursive function with a template parameter pack, like so:

```c++
template <typename Key>
JsonVariantConst resolvePath(JsonVariantConst variant, Key key) {
  return variant[key];
}

template <typename Key, typename... Tail>
JsonVariantConst resolvePath(JsonVariantConst variant, Key key, Tail... tail) {
  return resolvePath(variant[key], tail...);
}

template <typename T, typename... Keys>
T value(const char* json, Keys... keys) {
  DynamicJsonDocument doc(4096);
  deserializeJson(doc, json);
  return resolvePath(doc, keys...).template as<T>();
}

const char data[] = R"=( {"a": {"b": {"x2": 2}}})=";
int8_t x = value<int8_t>(data, "a", "b", "x2"); // 2
```

I know this looks complicated, but it is a common pattern.
BONUS: it works with arrays as well :-)

Demo: https://wandbox.org/permlink/PgICTZ1nIjAULTKO

Best regards,
Benoit

sorokolokarim, 2021-02-21:
Hi @bblanchon 🥇
Thank you very very very much !!! Tu me sauves enormement avec cette solution ! You are a master !! I hope on day get your level or more  :)

closed 2021-02-24 by sorokolokarim