bblanchon/ArduinoJson #1506: DeserializationError = Invalid Input

SalvoVhett, 2021-02-21:
I am working a project using an ESP8266. The code was several years old and I had to update to Json 6. I'm very new to coding. When I run the code I get an Invalid Input error. I believe this is because the GET command returns a response with headers that are not being removed. As shown by the response of the date in the result below.  If I hard code the payload info the code works.

    String url = "/api/emotions/primary/totals";
    Serial.print("requesting URL: ");
    Serial.println(url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
      "Host: " + host + "\r\n" +
      "User-Agent: BuildFailureDetectorESP8266\r\n" +
      "Connection: close\r\n\r\n");
    
    Serial.println("request sent");
      while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "r") {
        Serial.println("headers received");
      break;
      }      
    }
    
    String line = client.readStringUntil('\n');
    
    Serial.println("reply was:");
    Serial.println("==========");
    Serial.println(line);
    Serial.println("==========");
    Serial.println("closing connection");

    char * response = new char[line.length() + 1];

    strcpy(response,line.c_str());

    //char json[] = "{\"love\":2194363,\"surprise\":1926165,\"other\":995344501,\"joy\":17661349,\"*\":29013287,\"sadness\":5287209,\"anger\":1529641,\"fear\":1115696}";
  
    StaticJsonDocument<200> doc;

    DeserializationError error = deserializeJson(doc, response);
       if (error) {
       Serial.print("deserializeJson() failed with code ");
       Serial.println(error.c_str());
       
  } 

Result is
......connecting to wefeel.csiro.au
requesting URL: /api/emotions/primary/totals
request sent
reply was:
==========
Date: Sun, 21 Feb 2021 01:26:36 GMT

==========
closing connection
deserializeJson() failed with code InvalidInput
TweetCount
0
0
0
0
0
0
0
0
no update

How can I remove the headers so I only get the payload or body or the response?

bblanchon, 2021-02-21:
Hi @SalvoVhett,

I'm not sure how I can help you besides pointing you to the documentation:

* [DeserializationError::InvalidInput](https://arduinojson.org/v6/api/misc/deserializationerror/#invalidinput)
* [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/)
* [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/)
* BONUS: [How to format strings without the String class](https://cpp4arduino.com/2020/02/07/how-to-format-strings-without-the-string-class.html)

Best regards,
Benoit

SalvoVhett, 2021-02-21:
Yes, thanks,  the helped a lot. The get response had a few extra characters at the beginning. The Tutorial and Example code helped me arrive a a solution. Not very pretty but it works, I'll keep learning and see if I find a more elegant way. In the mean time my project works : )

bblanchon, 2021-02-22:
> The get response had a few extra characters at the beginning

This sounds like [Chunked transfer encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).
Did you try to use `HTTP/1.0` instead of `HTTP/1.1`?

SalvoVhett, 2021-02-25:
I added an extra line of code to read the extra chars and get rid of them.
I'll look into using http 1.0

bblanchon, 2021-03-19:
@SalvoVhett, any progress?

SalvoVhett, 2021-03-20:
Yep, It works great! Thanks for the help

closed 2021-03-20 by bblanchon