bblanchon/ArduinoJson #1508: Deserialize from reference of another JsonDocument.

gnalbandian, 2021-02-23:
Hi @bblanchon. Hope you're doing fine.

I'm having the following issue, or at least, request for help.

I'm receiving a json string which I deserialize just fine. In it, there at key that holds another string that happens to be another json.
ie:
`{
	"inline_keyboard": [
		[{
			"text": "Click to Override",
			"callback_data": "{\"Overrides\":{\"id\":1,\"alarms\":[2]}}"
		}]
	]
}`

Once the first deserialization is successfully done, I've got a JsonObject "callback_data" which holds a string (json string).

I need to deserialize that one too.
My first attempt was to create another JsonDocument and deserialize over it, from the reference of the  JsonObject "callback_data". This didn't work.

`
StaticJsonDocument<256> doc;
StaticJsonDocument<256> doc2;
deserializeJson(doc, completeMessage);

JsonObject callback_data = doc["inline_keyboard"][0][0]["callback_data"];

deserialize(doc2, callback_data);
`

I ended up creating a String variable to hold the callback_data content, and deserialize it from there
ie:
`
StaticJsonDocument<256> doc;
StaticJsonDocument<256> doc2;
deserializeJson(doc, msg.replyMarkup);

String data = doc["inline_keyboard"][0][0]["callback_data"].as<String>();
                        
deserializeJson(doc2, data);
`

It works, but, it seems to me redundant to create a new string holding the same content that's already present at the first JsonDocument.

Please, let me know what you think about this, and what would be your approach to solve this.
Regards!

Gonzalo

bblanchon, 2021-02-24:
Hi Gonzalo,

As you can see from [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) documentation, since v6.15.2, it supports an overload taking a [`JsonVariantConst`](https://arduinojson.org/v6/api/jsonvariantconst/), but not a [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/).

This means you can fix your program by changing the type of  `callback_data` from `JsonObject` to `JsonVariantConst`.
However, you can go one step further and eliminate this variable,  like so:

```c++
deserialize(doc2, doc["inline_keyboard"][0][0]["callback_data"]);
```

Demo: https://wandbox.org/permlink/oKORnqoixtyIceAb

Best regards,
Benoit

bblanchon, 2021-02-24:
PS: I added [a dedicated example](https://arduinojson.org/v6/api/json/deserializejson/#deserialize-an-object-member) in the documentation.

gnalbandian, 2021-02-24:
So simple! Thanks @bblanchon.
I appreciate very much your help and effort.

Regards!
Gonzalo

closed 2021-02-24 by gnalbandian

bblanchon, 2021-02-25:
You're welcome, Gonzalo.
Thank you for using ArduinoJson; don't forget to [star the project](https://github.com/bblanchon/ArduinoJson/stargazers) to show your support :wink:
Good luck with your project!