bblanchon/ArduinoJson #1510: Possible bug? Null value after sending data

logon84, 2021-02-26:
Hi. I am working a in a pulse/edge counter to measure waterflow using a proximity sensor using and ESP-01 (ESP8266). I send to my mqtt broker the edges, pulses, bootdatetime and last edge datetime. Both bootdatetime and last_edge_datetime are converted from linux epoch time to human readable time with ctime function. The problem I am facing is that the edgedatime value that is sent becomes sometimes "null". After some debugging I discovered that the problem is not the ctime function or the function I built to get the epoch time, but the JsonDocument. This is my code:
``/***************************************************************************************** 
*
* Waterflow sensor (Pulse counter) V3.0
*
* Written in Arduino IDE.
* This counter uses interrupts on ESP-01's GPIO3.
* It is "talk-only" client. 
* It sends out data while pulses detected.
* NTP sync.
* It keeps adding pulses until poweroff/reset.
*
* MQTT Message consists of 
*       pulses - number of pulses counted since last device boot.
*       liters - pulses converted to liters.
*       edges - number of rising and falling edges detected. Useful to check if disc is moving comparing old and actual value.
*       moved_last_half_hour - '1' if a transition in input pin has been detected in the last half hour.
*       inactive_for_48hrs - '1' if no new data in last 48 hours. Good to alarm if sensor is not registering data for a long period.
*       last_edge - Last edge detected datetime.
*       since - Last booting datetime.
*
* Copyright (C) 2021 Rubén López <rubenlogon@yahoo.es>
* 
*
* This library is free software. You may use/redistribute it under The MIT License terms. 
*
*****************************************************************************************/
#include <Arduino.h>
#include <ArduinoJson.h>
#include <time.h>
#include <TZ.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include "waterflow_sensor.h"
#ifdef OTA
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>
#endif

bool send_data = true;
bool moved_in_last_half_hour = false;
bool moved_in_last_48hrs = false;
time_t Bootdatetime;
time_t Edgedatetime;

volatile bool edge_detected = false;
volatile bool pulse_reference; //indicates which kind of edge is going to be the reference for 1 pulse rev. (true = rising edge, false = falling edge)
volatile unsigned int Pulses = 0;
volatile unsigned int Edges = 0;
volatile unsigned long LastEdgeMillis;

void mqttCallback(char* topic, byte* payload, unsigned int length) {
  //Not used
}

WiFiClient WifiClient;
PubSubClient MqttClient(mqtt_broker, mqtt_port, mqttCallback, WifiClient);
StaticJsonDocument<192> payload;

void ICACHE_RAM_ATTR pulseHandler() {
  if(Edges == 0) {
    pulse_reference = bool(!digitalRead(PULSE_PIN));
  }
  if((unsigned long)(millis() - LastEdgeMillis) >= DEBOUNCE_MS) {
    edge_detected = true;
    Edges = Edges + 1;
    LastEdgeMillis = millis();
    if(pulse_reference == bool(digitalRead(PULSE_PIN))) { //We got a pulse (1 rev)
      Pulses = Pulses + 1;
    }
  }
}

time_t get_epoch_time() {
  time_t tnow = 0;
  tnow = time(nullptr);
  while (tnow<100000 || tnow == NULL) {
    tnow = time(nullptr);
    delay(100);
  }
  return tnow;
}

String get_formatted_time(time_t atime) {
  char* str_out;
  str_out = ctime(&atime);
  str_out[strlen(str_out)-1] = '\0';
  return str_out;
}

bool pubdata(void) {
  payload["pulses"]               = Pulses; // = disc revs
  payload["liters"]               = Edges*1000*WATERMETER_RESOLUTION_M3*(10/2); 
  payload["edges"]                = Edges;
  payload["moved_last_half_hour"] = int(moved_in_last_half_hour);
  payload["inactive_for_48hrs"]   = int(!moved_in_last_48hrs);
  payload["let"]                  = ctime(&Edgedatetime)[15] - 48; //for debug purposes
  if(Edges == 0){
    payload["last_edge"]            = "-";
  }
  else{
    payload["last_edge"]            = ctime(&Edgedatetime);
  }
  payload["since"]                = ctime(&Bootdatetime);

  char buffer[512];
  serializeJson(payload, buffer);
    
  if(!MqttClient.publish(dest_topic, buffer, true)) {
    //Fail to publish
    if(!MqttClient.connected()) {
      //Not connected to broker
      if(!MqttClient.connect(client_id, mqtt_username, mqtt_password)) {
        //Can't connect to broker - Maybe wifi down? restart loop
      }
    }
    return false;
  }
  else {
    return true;
  }
}

void setup() {
  //Serial.begin(115200, SERIAL8N1, SERIALTX_ONLY);
  pinMode(blueLedPin, OUTPUT);
  digitalWrite(blueLedPin, HIGH);
  pinMode(PULSE_PIN, INPUT);
  attachInterrupt(digitalPinToInterrupt(PULSE_PIN), pulseHandler, CHANGE);

  WiFi.hostname(client_id);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    // wait 500ms, flashing the blue LED to indicate WiFi connecting...
    digitalWrite(blueLedPin, LOW);
    delay(250);
    digitalWrite(blueLedPin, HIGH);
    delay(250);
  }
  MqttClient.connect(client_id, mqtt_username, mqtt_password);
  configTime(TIME_ZONE, "es.pool.ntp.org");
  yield();
  Bootdatetime = get_epoch_time();
#ifdef OTA
  ArduinoOTA.setHostname(client_id);
  ArduinoOTA.setPassword(ota_password);
  ArduinoOTA.begin();
#endif
}

void loop() {
  if(WiFi.status() == WL_CONNECTED) {
    if((unsigned long)(millis() - LastEdgeMillis) >= LED_FLICKER_MS){
      digitalWrite(blueLedPin, LOW); //There's wifi, then led ON
    }
    if(moved_in_last_half_hour && (unsigned long)(millis() - LastEdgeMillis) >= 1800 * 1000) {
      moved_in_last_half_hour = false;
      send_data = true;
    }
    if(moved_in_last_48hrs && (unsigned long)(millis() - LastEdgeMillis) >= 3600 * 24 * 2 * 1000) {
      moved_in_last_48hrs = false;
      send_data = true;
    }
    if(edge_detected) {
      moved_in_last_half_hour = true;
      moved_in_last_48hrs = true;
      Edgedatetime = get_epoch_time();
      digitalWrite(blueLedPin, HIGH); //turnoff led
      send_data = true;
      edge_detected = false;
    }
    if(MqttClient.connected()) {
      if(send_data && (unsigned long)(millis() - LastEdgeMillis) >= MIN_TIME_BETWEEN_SENDS_MS && pubdata()) {
        send_data = false;
      }
      MqttClient.loop();
    }
    else {
      if(MqttClient.connect(client_id, mqtt_username, mqtt_password)) {
        MqttClient.loop();
      }
    }
  } 
  else {
    digitalWrite(blueLedPin, HIGH); //No wifi, no led light
    WiFi.begin(ssid, password);
    while(WiFi.waitForConnectResult() != WL_CONNECTED);
  }
#ifdef OTA
  ArduinoOTA.handle();
#endif
}``

Let's focus on the pubdata function. I added the "let" key  for debug purposes (Last Edge Time). ctime(&Edgedatetime)[15]  returns character 15 of the ctime output which is exactly the value of the minute the Lastedgedatetime was detected (minus 48 to convert int to actual char value). Now this screeshot was taken just after receiving a lat edge on my device:
https://imgur.com/a/fE3ZXK3

As you can see in the screenshot, ctime in fact returned some data because let key rturns the correct minute of time. But "Last edge" gets a null value. As I said, not always....I calculated the size of the jsondocument using your v6 online tool and so I set the 192 value. The Json library I am using in Arduino is 6.17.3. Any help would be appreciated. Thanks

bblanchon, 2021-02-27:
Hi @logon84,

This is not exactly an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example), so I cannot give you a definite answer, but I'll try my best to give a relevant response.

I think the issue is in the return type of `ctime()`: `char*`
Indeed, [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) treats `const char*` and `char*` differently:

* `const char*` are stored by pointer (so no additional space is required)
* `char*` are stored by copy (so they take n+1 extra bytes in the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/))

Note that the [ArduinoJson Assitant](https://arduinojson.org/v6/assistant/) has two checkboxes in the "Tweaks" part of Step 3.

There are two ways to fix this problem:

1. Cast the result of `ctime()` to `const char*` (the type it should have been in the first place)
2. Increase the capacity of the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) to make some room for the string.

But that's not all! There is a second problem in your program: it reuses the same [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) without resetting it, which creates a leak.

Again, there are two ways to fix this problem:

1. Move the declaration of `payload` inside the function (highly recommend)
2. Call [`JsonDocument::clear()`](https://arduinojson.org/v6/api/jsondocument/clear/)

Lastly, remember that, by default, the maximum payload allowed by PubSubClient is 256 bytes, so allocating a 512-bytes buffer is useless.

See also:

* [Why is the output incomplete?](https://arduinojson.org/v6/issues/incomplete-output/)
* [I found a memory leak in the library!](https://arduinojson.org/v6/issues/memory-leak/)
* [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)

Best regards,
Benoit

logon84, 2021-02-27:
Hi, thanks for your detailed reply. I think I solved the problem with your advices. Thanks again!

closed 2021-02-27 by logon84

bblanchon, 2021-02-28:
You're welcome, @logon84.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: