bblanchon/ArduinoJson #1511: Pass JasonObject as parameter and return from function

Kasperdelasopa, 2021-02-28:
Hallo,

i'm struggling with receiving a JsonObject from a HelperCall and pass it to an other function as a Parameter.

I try to do something like this:

```
JsonObject obj= plan->getJsonObject();
Serial.printf("Size1 %d\n", obj.size());
_config->persist(obj);
```

The persist function looks like this:

```
bool Config::persist(JsonObject obj){
   JsonObject lobj = obj;
   Serial.printf("Size2 %d\n", lobj.size());
   ...
}
```

The output is: "Size1 1" and the second one: "Size2 0". Therfore the JsonObject before i pass it to the persist() function is not empty. directly after i passed it as a parameter, the object is empty. 

I'm sure if missed the designated way to do this. Can someone tell me what i'm doing wrong.

bblanchon, 2021-03-01:
Hi @Kasperdelasopa,

There doesn't seem to be anything wrong with the code above.
I think the problem is in `getJsonObject()`: shouldn't it return a `DynamicJsonDocument` instead?

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

bblanchon, 2021-03-01:
PS: if you don't manage to solve this issue by returning a `JsonDocument`, please write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example).

Kasperdelasopa, 2021-03-01:
> I think the problem is in `getJsonObject()`: shouldn't it return a `DynamicJsonDocument` instead?

Is is not possible to receive a JsonObject from a function as i did? 
Actually i replaced it to DynamicJsonDocument and it worked.

Are there limitations by receiving a JsonObject and pass that object to an other function as an argument?

bblanchon, 2021-03-03:
[JsonObject](https://arduinojson.org/v6/api/jsonobject/) is a reference to an object stored in a [JsonDocument](https://arduinojson.org/v6/api/jsondocument/); it cannot exist without the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/).

Kasperdelasopa, 2021-03-14:
Thx allot, this helped me quite much! 

In my case, it seemed that i losed the original reference of the JsonObject, after a scope change.
Now i have worked with that in mind.

An it seems to work, when i need it.

But i recognized, that calling a function creating a bigger DynamicJsonDocument (8192b) multiple times makes the object to not accept the data anymore. It seems like that the Object are not always destroyed and reallocated correctly.

The times i can call a particular function before the data in the object is not accessible anymore seems to directly depends on the size of the created DynamicJsonDocument.

In my case I'm using the Arduino library on a ESP8266 is there something special, to deal with on that controller?

May i need to change the free() function?

Did someone experienced the same issue. I'm sure its just my lack of experience C++ which cases this maybe someone has an idea.

Best regards and thx again for the help so far!

bblanchon, 2021-03-15:
Hi @Kasperdelasopa,

Please read:

* [I found a memory leak in the library!](https://arduinojson.org/v6/issues/memory-leak/)

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

Kasperdelasopa, 2021-03-15:
Thx for the link.

I was already aware of these page.

Actually i only use DynamicJsonDocuments as the most of my JSON Object are bigger than 1kb.
In addition all used DynamicJsonDocuments are creation within functions which are called.
Therefore i would expect that all DynamicJsonDocuments within the function are destroyed/freed once the code leaves the function scope and returns to the calling scope.

For these Documents there is only one exception, where i use a global DynamicJsonDocuments.
But every time, when i change an object within this global Document, i call garbageCollect();

In addition i started to call .clear() for all  the other local scoped DynamicJsonDocuments, because i got desperate on this issue.
The result was, that i was able to call the respective functions more often before i was not able to allocate additional DynamicJsonDocuments within further function calls.

In the Link you provided, was mentioned that using the String class might be bad as well, so i switched to only use (const) char*

Could there be anything which can prevents the memory of my DynamicJsonDocuments to be fully reclaimed after leaving a function scope?

bblanchon, 2021-03-17:
It's hard to tell exactly what's wrong without looking at your code and without even knowing the type of hardware it's running on, but this certainly sounds like a fragmentation issue.
Removing `String` is a good thing, but it may not be enough.
Would you mind sharing a bit more information about your project?

Kasperdelasopa, 2021-04-02:
Hi,

sorry for beeing quite so long, but i haven't much time for my hobbies in the last weeks.

My Project is about a device which i want to control with a schedule i provide as JsonObjects. These JsonObjects are representing time intervals which, for example, represent a temperature, which is intended for that time period.
The list of time intervals are provided as as JsonArray.

That works awesome, as long as these configuration is provides over SD-Card and and rad it from there once at device start.

In addition i implemented a REST API to update the time intervals. --> Here is where my problems starts.

Having a bigger list of time intervals, the JsonDocument i need to provide is about 4kb of data.

I can perform an update about 3-4 times before the memory leak happens.

Following your Link (I found a memory leak in the library!) i invested a big amount of effort to try to eliminate the usage if the String class. Unfortunate i'm not much into the details of C++ (not much experience). This is why i thought std::string will do the job without creating memory leaks --> I learned that this is not the case.  That's why i'm on the starting point again, where i'm not able to get rid of the memory leak even if there is no change on a DynamicJsonDocument without calling garbageCollect();, but still creating memory leaks

bblanchon, 2021-04-10:
Again, it's hard to give any advice without looking at your code.
Also, I didn't say there was a memory leak; I said there was probably a [memory fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html) issue.

Kasperdelasopa, 2021-04-11:
I'm totally with you, but the code is spreaded over multiple files and not the best written one.

Therefore i might need to live with that.

Now i'm also sure its memory fragmentation caused by to much std::string and String-Class operations.

But Thank your for your help so far!! 
at least i know where the issue comes from.

closed 2021-04-11 by Kasperdelasopa

bblanchon, 2021-04-12:
You're welcome, @Kasperdelasopa,
Good luck with your project!