bblanchon/ArduinoJson #1512: Getting issue when use Arduinojson with ESP32 (Board: ESP32 Dev Module) with WIFI

mwaseemwali, 2021-03-01:
whenever i use Arduino JSON library with ESP32, Dev Module board. my wifi disconnect after a little time passed.

vortex314, 2021-03-01:
No Issue here, I used ArduinoJson on ESP32 with esp32_idf and the device keeps running for weeks. Just be aware that JsonDocument could need clear() after use.
See : https://arduinojson.org/v6/api/jsondocument/clear/

bblanchon, 2021-03-02:
Hi @mwaseemwali,

This could be due to slow serialization.
Please read:

* [How to improve (de)serialization speed?](https://arduinojson.org/v6/how-to/improve-speed/)

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

ratzi30, 2021-03-12:
@mwaseemwali, 
I had the same problem with my ESP32.
Check if you have a memory leak, this will cause the wifi to disconnect.

closed 2021-03-13 by bblanchon