bblanchon/ArduinoJson #1513: Backwards compatibility issue ? no matching function for call to getOrAddUpstreamMember().set(src);

vortex314, 2021-03-01:
```
void commandArguments(JsonObject config, int argc, char **argv) {
  int opt;

  while ((opt = getopt(argc, argv, "f:m:l:v:")) != -1) {
    switch (opt) {
      case 'm':
        config["mqtt"]["connection"] = optarg;
        break;
      case 'f':
        config["configFile"] = optarg;
        break;
      case 'v': {
        char logLevel = optarg[0];
        config["log"]["level"] = logLevel;
        break;
      }
      case 'l':
        config["log"]["file"] = optarg;
```
../ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp: In instantiation of ‘typename ArduinoJson6173_71::enable_if<(! ArduinoJson6173_71::is_array<TValue>::value), ArduinoJson6173_71::MemberProxy<TParent, TStringRef>&>::type ArduinoJson6173_71::MemberProxy<TParent, TStringRef>::operator=(const TValue&) [with TValue = char; TObject = ArduinoJson6173_71::MemberProxy<ArduinoJson6173_71::ObjectRef, const char*>; TStringRef = const char*; typename ArduinoJson6173_71::enable_if<(! ArduinoJson6173_71::is_array<TValue>::value), ArduinoJson6173_71::MemberProxy<TParent, TStringRef>&>::type = ArduinoJson6173_71::MemberProxy<ArduinoJson6173_71::MemberProxy<ArduinoJson6173_71::ObjectRef, const char*>, const char*>&]’:
main.cpp:114:34:   required from here
../ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:63:5: error: no matching function for call to ‘ArduinoJson6173_71::VariantRef::set(const char&)’
   63 |     getOrAddUpstreamMember().set(src);
      |     ^~~~~~~~~~~~~~~~~~~~~~
The line is : 
```
        config["log"]["level"] = logLevel;
```

vortex314, 2021-03-01:
Adapted the code and works again : 
```
        std::string logLevel;
        logLevel += optarg[0];
        config["log"]["level"] = logLevel;
```

bblanchon, 2021-03-02:
Hi @vortex314,

Assuming you're talking about the master branch, then yes: the `char` type is not supported anymore.
This change was needed to fix #1498.
As explained in the [changelog](https://github.com/bblanchon/ArduinoJson/blob/54d4b308f4e839c7986c913f77b77be96d9b03fb/CHANGELOG.md), you can use an `int8_t` instead.

Best regards,
Benoit

vortex314, 2021-03-02:
Ok, adapted my code. Thanks for the follow-up.

closed 2021-03-02 by vortex314