bblanchon/ArduinoJson #1517: SpiffsConfig array

HilverinkJ, 2021-03-03:
Hi,

I'm trying to modify your example "SpiffsConfig" from the book. But i'm struggling to find the right solution the array.
This is the output

{
  "server": {
    "username": "admin",
    "password": "s3cr3t",
    "host": "example.com",
    "path": "/resource",
    "sensor": [
      null,
      null
    ]
  }
}
Done!

What have I done wrong?

[array.zip](https://github.com/bblanchon/ArduinoJson/files/6078463/array.zip)

bblanchon, 2021-03-04:
Hi @HilverinkJ,

Thank you very much for purchasing my book; your support means a lot!

From what I understood of your project, you need to replace the loop in `Config::save()` with:

```c++
for (int i = 0; i < elements; i++) {
  element[i].save(sen.createNestedArray());
}
```

This produces the following output:

```json
{
  "server": {
    "username": "admin",
    "password": "s3cr3t",
    "host": "example.com",
    "path": "/resource",
    "sensor": [
      [
        "VALUE 10",
        ""
      ],
      [
        "",
        "VALUE 20"
      ]
    ]
  }
}
```

As you can see, this perfectly mirrors your data structures.

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

HilverinkJ, 2021-03-04:
Hi Benoit,
I'm just a hobbyist and try to understand your accelent work. Im sorry that I not have made myself clear. My intention is to store as array.
` "sensor": [ "value_1" , "value_2" , "......" ] `

Regards,
Joop

bblanchon, 2021-03-19:
Hi Joop,

Did you find a solution to your problem?

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

HilverinkJ, 2021-03-27:
Hi Benoit,
No I did not.
This is the output I like to create:

`{`
`"server": {`
`     "host": "www.example.com",`
`     "path": "/upload",`
`     "username": "admin",`
`     "password": "secret",`
`     "sensor": [ `
`            "value_1",` 
`            "value_2",`
`            "value_3",` 
`            "value_3",` 
`            "value_4"`
`            ]`
`   }`
` }`
Regards,

Joop.

[array.zip](https://github.com/bblanchon/ArduinoJson/files/6216402/array.zip)

bblanchon, 2021-04-10:
Hi Joop,

Well, in that case, you need to change your `struct` to something like:

```c++
struct Config {
  static const int maxElements = 4;
  char element[maxElements][32];
  int elements = 0;
  ...
};
```

...and change `Config::save()` to something like:

```c++
for (int i = 0; i < elements; i++) {
  sen.add(element[i]);
}
```

This allows you to have up to 4 sensors in your configuration, each sensor having a name limited to 31 characters.

Best regards, 
Benoit

HilverinkJ, 2021-04-10:
Hi Benoit,

That did the trick. 
I was turning around in circels.

Thanks a lot.

Regards,
Joop.

closed 2021-04-10 by HilverinkJ

bblanchon, 2021-04-11:
You're welcome, Joop.
Good luck with your project!