bblanchon/ArduinoJson #1518: compilation Error using example JsonParserExample Wemos D1 R1 All versions

miguelangelwww, 2021-03-04:
When compiling JsonParserExample in the installed examples of any versión (installed right now 6.17)  in Arduino 1.8.7 (Mac OS 10.14.6) the following error appears 

n file included from /Users/usuario/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:259:0,
                 from sketch/JsonParserExample.ino.cpp:1:
/Users/usuario/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/pgmspace.h:16:51: error: __c causes a section type conflict with s0
 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
                                                   ^
/Users/usuario/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h:38:76: note: in definition of macro 'FPSTR'
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                                                                            ^
/Users/usuario/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h:39:34: note: in expansion of macro 'PSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                  ^
/Users/usuario/Documents/Arduino/libraries/ArduinoJson/examples/JsonParserExample/JsonParserExample.ino:44:18: note: in expansion of macro 'F'
     Serial.print(F("deserializeJson() failed: "));

                  ^
In file included from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp:14:0,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/convertNumber.hpp:18,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:9,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/usuario/Documents/Arduino/libraries/ArduinoJson/examples/JsonParserExample/JsonParserExample.ino:9:
/Users/usuario/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:93:5: note: 's0' was declared here
     ARDUINOJSON_DEFINE_STATIC_ARRAY(char, s0, "Ok");
     ^
exit status 1
Error compilando para la tarjeta WeMos D1 R1.

bblanchon, 2021-03-04:
Hi @miguelangelwww,

Thank you for reporting this issue.
I have a very similar setup on my computer and the program compiles correctly.
This error looks very similar to https://github.com/esp8266/Arduino/issues/5036; I don't think I can do anything on the library side.
As a workaround, you can disable the support of PROGMEM in ArduinoJson by setting [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to 0.

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

bblanchon, 2021-03-17:
@miguelangelwww, any update?

jjsuwa-sys3175, 2021-03-28:
some information:
- ESP8266 Arduino cores after 1st October 2019 use unique sections for each `PSTR()` calls, thus such conflict do not occur (see https://github.com/esp8266/Arduino/commit/97c926ea917049885df4ce57cc935062fccf8c50).

closed 2021-04-13 by bblanchon