bblanchon/ArduinoJson #1519: Using ArduinoJson V6 with Industruino controller

johnmacleod101, 2021-03-05:
I have a [Industruino controller ](https://industruino.com/shop/product/industruino-proto-d21g-8)with a MCU type: ATSAMD21G18 (Arduino Zero/MKR equivalent)

- I am using Arduino IDE version 1.8.13.

- I would like to use ArduinoJson version 6.17.3.

If I use V5 of ArduinoJSON library I can compile without error but using V6 I get the following error as soon as I add INCLUDE reference and compile.

**Below is error on compile.**

_Compiling sketch...
/Users/johnmacleod/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_INDUSTRUINO_D21G -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2E78 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"ES Gear Ltd.\"" "-DUSB_PRODUCT=\"Industruino D21G\"" -I/Users/johnmacleod/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/johnmacleod/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.0.0/CMSIS/Device/ATMEL/ -I/Users/johnmacleod/Library/Arduino15/packages/industruino/hardware/samd/1.0.1/cores/industruino -I/Users/johnmacleod/Library/Arduino15/packages/industruino/hardware/samd/1.0.1/variants/industruino_d21g -I/Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src -I/Users/johnmacleod/Documents/Arduino/libraries/PubSubClient/src -I/Users/johnmacleod/Documents/Arduino/libraries/Adafruit_SleepyDog_Library -I/Users/johnmacleod/Documents/Arduino/libraries/MCP7940/src -I/Users/johnmacleod/Library/Arduino15/packages/industruino/hardware/samd/1.0.1/libraries/Wire -I/Users/johnmacleod/Library/Arduino15/packages/industruino/hardware/samd/1.0.1/libraries/SPI -I/Users/johnmacleod/Documents/Arduino/libraries/Ethernet2-master/src -I/Users/johnmacleod/Documents/Arduino/libraries/Indio/src -I/Users/johnmacleod/Documents/Arduino/libraries/DallasTemperature -I/Users/johnmacleod/Documents/Arduino/libraries/OneWire -I/Users/johnmacleod/Documents/Arduino/libraries/UC1701/src /var/folders/pc/k007fj356nd9w4snbzm834_c0000gn/T/arduino_build_741529/sketch/UFH_Controller_5thMarch2021Industruino.ino.cpp -o /var/folders/pc/k007fj356nd9w4snbzm834_c0000gn/T/arduino_build_741529/sketch/UFH_Controller_5thMarch2021Industruino.ino.cpp.o
In file included from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/static_array.hpp:11:0,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp:14,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/convertNumber.hpp:18,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:9,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/johnmacleod/Documents/Arduino/UFH_Controller_5thMarch2021Industruino/UFH_Controller_5thMarch2021Industruino.ino:1:
/Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6173_91::enable_if<ArduinoJson6173_91::is_pointer<T>::value, T>::type ArduinoJson6173_91::pgm_read(const void*) [with T = const __FlashStringHelper*; typename ArduinoJson6173_91::enable_if<ArduinoJson6173_91::is_pointer<T>::value, T>::type = const __FlashStringHelper*]':
/Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:103:12:   required from here
/Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:45: error: 'const void*' is not a pointer-to-object type
   return reinterpret_cast<T>(pgm_read_ptr(p));
                                             ^
Using library ArduinoJson at version 6.17.3 in folder: /Users/johnmacleod/Documents/Arduino/libraries/ArduinoJson 
Using library PubSubClient at version 2.8 in folder: /Users/johnmacleod/Documents/Arduino/libraries/PubSubClient_ 

I know there were a lot of changes to V6 library but any clue as to starting to find out what problem is would be great.
Thanks

bblanchon, 2021-03-05:
Hi @johnmacleod101,

Thank you for reporting this problem.

> error: 'const void*' is not a pointer-to-object type

This is the same error we encountered in #1381.
It was caused by a bug in Arduino Core API, which was fixed by arduino/ArduinoCore-API#118 in October.

Make sure the Arduino core for your platform is up-to-date.
If the problem still occurs, please contact the authors of the core and tell them about arduino/ArduinoCore-API#118.

As a workaround, you can disable `PROGMEM` support in the library by setting [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`.

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

johnmacleod101, 2021-03-05:
Hi @bblanchon 
Thanks for your prompt reply. I tried updating the core without success. I am now at 1.03 previously was at 1.01.
But your advice of disabling PROGMEM worked.
I will contact the Core provider for the Industruino core for SAMD21 and make them aware of the issue.
Thanks again for your prompt reply.

closed 2021-03-05 by johnmacleod101