bblanchon/ArduinoJson #1520: Problem with delay() after deserializeJson()

ratzi30, 2021-03-12:
Hello, 

I want to send a JSON document from an ESP32 to an Arduino via hardware Serial.
My cleaned code looks like this:
```
#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200);
  Serial2.begin(19200);
}

void loop() {
  if (Serial2.available()) {
    
    DynamicJsonDocument doc(512);

    DeserializationError errorJson = deserializeJson(doc, Serial2);

    if (errorJson) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(errorJson.f_str());
      return;
    }
    
    delay(200);

    serializeJson(doc, Serial);
    Serial.println();
  }
}
```
The `delay(200)` is representative of other parts of my program.
If I remove this delay, everything works fine and the output looks like this:
`{"PWM1":0,"PWM2":0,"PWM3":0,"steeringSpeed":67,"steeringAngle":48,"ShuttleLever":"rest"}`

But with the delay (or some functions) in the code, the JSON will not be parsed correctly:
[image]

I have tested a lot and no clue what could be wrong.
Thanks in Advance!

bblanchon, 2021-03-13:
Hi @ratzi30,

Thank you for this well-structured question 👍 

Since many users will benefit from the answer, I decided to add a [dedicated section in How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/#incompleteinput).

Here is the relevant part:

> In the first case, the sender writes bytes faster than the receiver reads them, filling up the receiver’s serial buffer until it overflows and drops incoming bytes. For example, **this problem happens when the receiver is busy doing some other task when the sender is transmitting**. It also occurs when the receiver logs incoming data at a slower rate; that’s why it’s crucial to get the “debug” serial running much faster than the “communication” serial.
>
> You could solve this issue by increasing the serial buffer size; the details depend on each platform. **For Arduino UNO, the default is 64 and can be changed by defining the `SERIAL_RX_BUFFER_SIZE` macro**. For ESP8266 and ESP32, the default is 256 and can be changed by calling setRxBufferSize().

If you use the Arduino IDE, you'll see that defining `SERIAL_RX_BUFFER_SIZE` is not as easy as it should; I let you figure out the details.

BTW, remember that the UNO is 5V whereas the ESP8266 is 3.3V, so you may need a logic level adapter.
I read on several sites that the ESP8266 is 5V tolerant, but I don't know for sure.

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

ratzi30, 2021-03-13:
Thanks @bblanchon, for the very fast answer!

Changing the `SERIAL_RX_BUFFER_SIZE` worked perfectly.
About the voltage levels ... I'm working with an ESP32 which is based on 5V logic ;)

In case anyone who reads this,  would like to change the buffer size on the Arduino mega 2560, it can be found in:
...\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino\HardwareSerial.h

closed 2021-03-13 by ratzi30