bblanchon/ArduinoJson #1521: ESP32 nested iterator with multiple duplicate key values question

frankcohen, 2021-03-20:
Thank you for all your work on ArduinoJson - it is excellent. I am defining my own JSON structure to control a video playback show entertainment system. It runs on ESP32 and I am building the code in Arduino IDE 1.8.13. The structure is:

```
{
	"ReflectionsShow": {
		"title": "Candle blowing",
		"events": {
			"event": {
				"Name": "First try",
				"trigger": "OnStart",
				"comment": "Plays a sequence once upon startup",
				"sequence": {
					"PlayAtOnce": {
						"comment": "PlayAtOnce plays from the same starting time, and stops after all ends",
						"playvideo": "Candle.mpg",
						"playaudio": "Tron.m4a"
					},
					"PlayAtOnce": {
						"playvideo": "Hurry.mpg",
						"playaudio": "Hurry.m4a"
					}
				}
			}
		}
	}
}
```
The **sequence** key contains one or more PlayAtOnce keys. When I use this:

`JsonObject sequence = doc["ReflectionsShow"]["events"]["event"]["sequence"];
`
I get sequence.size() = 1. I expect this to be 2 as there are two "PlayAtOnce" keys.

I am needing a way to iterate through an unknown set of nested keys. Hopefully using something as simple as this:

```
  JsonObject seqs = doc["ReflectionsShow"]["events"]["event"]["sequence"];

  for (auto sequence : seqs){

    if ( strcmp( sequence, "PlayAtOnce" )
    {
      //      
    }

    if ( strcmp( sequence, "PlayAtEnd" )
    {
      //      
    }
      
    //
  }
```
What do you recommend?
-Frank

frankcohen, 2021-03-21:
A more basic question: Does ArduinoJson 6 support duplicate keys?

frankcohen, 2021-03-21:
Found your page talking about duplicate keys at https://github.com/bblanchon/ArduinoJson/issues/1095, "So deserializeJson() should return Ok, but the JsonDocument should contain the last value."

"Do you have a use case where detecting duplicate keys is crucial?" I am created an entertainment product that runs video and audio shows. I designed a JSON structure as a procedural scripting language for the show elements (an example is above). If I stay with JSON I will have to serial number the PlayOnce keys to PlayOnce1, PlayeOnce2, etc.

Seems like a data-destructive bug for any parser to keep only the last duplicate key in a structure. There is no restriction against duplicate keys in the JSON standard. Oh well.

frankcohen, 2021-03-21:
I changed the JSON document to avoid duplicate keys:

```
{
	"ReflectionsShow": {
		"title": "Candle blowing",
		"version": "1",
		"events": {
			"1": {
				"type": "event",
				"name": "First try",
				"trigger": "OnStart",
				"comment": "Plays a sequence once upon startup",
				"sequence": {
					"1": {
						"playvideo": "Candle.mpg",
						"playaudio": "Tron.m4a"
					},
					"2": {
						"playvideo": "Hurry.mpg",
						"playaudio": "Hurry.m4a"
					}
				}
			},
			"2": {
				"type": "event",
				"name": "Left button pressed",
				"trigger": "LeftButtonPress",
				"comment": "Plays when left button pressed",
				"sequence": {
					"1": {
						"playvideo": "Candle.mpg",
						"playaudio": "Tron.m4a"
					},
					"2": {
						"playvideo": "Hurry.mpg",
						"playaudio": "Hurry.m4a"
					}
				}
			}
		}
	}
}
```
I wrote this code to parse the values:

```
  JsonObject events = doc["ReflectionsShow"]["events"];
  if ( events.isNull() )
  {
    Serial.println( "Events not found");
  }

  Serial.println( "@" );

  for (JsonObject::iterator it=events.begin(); it!=events.end(); ++it) {

      String eventskey = it->key().c_str();
      String eventsvalue = it->value().as<char*>();
      Serial.print( "Event " );
      Serial.println( eventskey );
      
      JsonObject event = doc["ReflectionsShow"]["events"][ eventskey ];

      String eventname = event["name"].as<char*>();

      Serial.print( "name " );
      Serial.println( eventname );

      JsonObject sequence = doc["ReflectionsShow"]["events"][eventskey]["sequence"];

      for (JsonObject::iterator sequenceit=sequence.begin(); sequenceit!=sequence.end(); ++sequenceit) {

          String seqkey = sequenceit->key().c_str();
          String seqvalue = sequenceit->value().as<char*>();

          Serial.print( "seqkey " );
          Serial.println( seqkey );

          JsonObject play = doc["ReflectionsShow"]["events"][eventskey]["sequence"][seqkey];

          for (JsonObject::iterator playit=play.begin(); playit!=play.end(); ++playit) {

            String playkey = playit->key().c_str();
            String playvalue = playit->value().as<char*>();
            
            Serial.print( playkey );
            Serial.print( " : " );
            Serial.println( playvalue );

          }          
        }
      }
```
 I get this output:
```
Event 1
name First try
seqkey 1
playvideo : Candle.mpg
playaudio : Tron.m4a
seqkey 2
playvideo : Hurry.mpg
playaudio : Hurry.m4a
Event 2
name Left button pressed
seqkey 1
playvideo : Candle.mpg
playaudio : Tron.m4a
seqkey 2
playvideo : Hurry.mpg
playaudio : Hurry.m4a
```
This works for me. I'd love to know your optimizations to the code. Thanks for a great library. -Frank

bblanchon, 2021-03-21:
Hi Frank,

I don't understand why you use an object instead of an array to store a sequence.
The order of the members in a JSON object is not guaranteed; it turns out that the current version of ArduinoJson returns them in the same order, but a future version could (and most likely will) return them in alphabetical order.

If I were you, I would use the following JSON document:

```json
{
  "ReflectionsShow": {
    "title": "Candle blowing",
    "version": "1",
    "events": [
      {
        "name": "First try",
        "trigger": "OnStart",
        "comment": "Plays a sequence once upon startup",
        "sequence": [
          {
            "playvideo": "Candle.mpg",
            "playaudio": "Tron.m4a"
          },
          {
            "playvideo": "Hurry.mpg",
            "playaudio": "Hurry.m4a"
          }
        ]
      },
      {
        "name": "Left button pressed",
        "trigger": "LeftButtonPress",
        "comment": "Plays when left button pressed",
        "sequence": [
          {
            "playvideo": "Candle.mpg",
            "playaudio": "Tron.m4a"
          },
          {
            "playvideo": "Hurry.mpg",
            "playaudio": "Hurry.m4a"
          }
        ]
      }
    ]
  }
}
```

Or, if by design there can be only one sequence per trigger, I would use:

```json
{
  "ReflectionsShow": {
    "title": "Candle blowing",
    "version": "1",
    "events": {
      "OnStart": {
        "name": "First try",
        "comment": "Plays a sequence once upon startup",
        "sequence": [
          {
            "playvideo": "Candle.mpg",
            "playaudio": "Tron.m4a"
          },
          {
            "playvideo": "Hurry.mpg",
            "playaudio": "Hurry.m4a"
          }
        ]
      },
      "LeftButtonPress": {
        "name": "Left button pressed",
        "comment": "Plays when left button pressed",
        "sequence": [
          {
            "playvideo": "Candle.mpg",
            "playaudio": "Tron.m4a"
          },
          {
            "playvideo": "Hurry.mpg",
            "playaudio": "Hurry.m4a"
          }
        ]
      }
    }
  }
}
```

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

frankcohen, 2021-03-21:
Thank you Benoit! The second proposed JSON schema works fine for my application. I am implementing the code to support it.

"I don't understand why you use an object instead of an array to store a sequence."

I did not find a way to make JsonArray work. It's probably a recessive gene in me - or maybe not reading the manual in the right way. Here is what I tried:

```
          JsonArray array = doc["ReflectionsShow"]["events"][eventskey]["sequence"][seqkey].as<JsonArray>();          
          for(JsonVariant playit : array) {
            String playvalue = playit.as<char*>();

```
array is null. I'm not sure how to navigate the structure of a doc and use JsonArray to get keys and values from the structure?

The most difficult part of learning ArduinoJson was the difference in support classes between DynamicJsonDocument, JsonVariant, JsonObject, and JsonArray. For example, the docs clearly say I get a JsonArray from a DynamicJsonDocument. So I expect that I get a JsonArray from a JsonObject too. The docs tell me when I use a JsonArray it's iterator will give me a JsonVariant. So I try to get the key from the JsonVariant. I'm feeling confused.

-Frank

bblanchon, 2021-03-22:
Hi Frank,

I understand how this can be confusing, but on the other hand, ArduinoJson provides a tool that literally writes the code for you.
Did you try the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/)?  It shows how to write this loop!

The code from the Assistant extracts all members, so maybe it's confusing as well...
Here is a shorter version that only extracts the sequence:

```c++
JsonArray sequence = doc["ReflectionsShow"]["events"][eventskey]["sequence"];
for (JsonObject step : sequence) {
  const char* video = step["playvideo"];
  const char* audio = step["playaudio"];
  // ...
}
```

Demo: https://wandbox.org/permlink/V4BfCGFiEYgEB1v5

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

frankcohen, 2021-03-22:
I made these mistakes:

JsonArray array = doc["ReflectionsShow"]["events"][eventskey]["sequence"][seqkey].as<JsonArray>();
I thought using the doc[] (or doc.getMember(“”) technique would return a JsonObject, that’s why I added .as<JsonArray>() to the end. Not needed. By casting it I get a null value for array.

I thought the JsonArray initializer only accepted DynamicJsonDocument or StaticJsonDocument. It works fine with JsonObjects.

I thought that using the JsonObject iterator would give me the nested keys in the order they appear in the JSON structure. The standard doesn’t require the order to be preserved. And, the underlying parser implementations often use key/value dictionaries that also do not keep the order. The ArduinoJson solution is to use JsonArrays.

I didn’t expect ArduinoJson’s initializers to work so well with various different object types. That’s really cool. For example, it didn’t occur to me that I could do further value filtering when the value is an object: 
for (JsonObject step : sequence) {
  const char* video = step["playvideo"];

I thought using a JsonArray would lose the context of the key I was iterating through. For example, when I’m in this loop:
JsonArray array = doc.as<JsonArray>();
for(JsonVariant v : array) {
    Serial.println(v.as<int>());
}
I thought I would not be able to use v as a JsonObject to do further processing on that key and its nested values below.

The ArduinoJson Assistant is brilliant. Your documentation, the object implementation, and the Assistant are wonderful. I had my own misunderstandings (above) that kept me from being successful.

There is enough here of my discussion with you that I’d like to write and publish an article on using ArduionJson 6 for procedural event scripting. That’s my way of giving back to the project. I’ll run it by you for edits.

Thanks!

-Frank

frankcohen, 2021-03-22:
Hi, I wrote up my experience on Reddit. I used the draft capability. Would you please let me know your edits. Thanks! -Frank https://www.reddit.com/user/frankcohen/draft/d64a4b8e-8b46-11eb-b511-5a5391b2adf9

bblanchon, 2021-03-23:
Hi Frank,

Thank you for the kind words :-)

I noticed you used an old style loop for the events array:

```c++
for (JsonObject::iterator it=events.begin(); it!=events.end(); ++it) 
  {
      String eventskey = it->key().c_str();
      String eventsvalue = it->value().as<char*>();
```

You can use range for loop:

```c++
for (auto event : events) 
  {
      String eventskey = event.key().c_str();
      String eventsvalue = event.value().as<const char*>();
```

(note that I also changed `as<char*>()` to `as<const char*>()`; this is a breaking change coming in 6.18)

You don't need `strcmp()` to compare keys or values from ArduinoJson; they all support `==` and `!=`.

Best regards,
Benoit

frankcohen, 2021-03-23:
Thank you Benoit! I am incorporating your changes. I will publish on Reddit today. -Frank

frankcohen, 2021-03-23:
Published at https://www.reddit.com/r/arduino/comments/mbdj53/json_procedural_scripting_using_arduinojson_and/

closed 2021-04-13 by bblanchon