bblanchon/ArduinoJson #1522: How to serialize and deserialize array

t0maszduda, 2021-03-25:
I have array "data" with numbers:

#include <ArduinoJson.h>

double data[29] = {31,40,50,63,80,100,125,160,200,250,315,400,500,630,800,1000,1250,1600,2000,2500,3150,4000,5000,6300,8000,10000,12500,16000,20235};

and the question is how to serialize and deserialize this array as the same array on the second Arduino between this doc:

void loop() {

StaticJsonDocument<500> doc;
float t = bme.readTemperature();
doc["temp"] = t;
float h = bme.readHumidity();
doc["hum"] = h;
float p = bme.readPressure() / 100.0F;
doc["pre"] = p;
serializeJson(doc, s);

}

bblanchon, 2021-03-26:
Hi @t0maszduda,

Have a look at [copyArray()](https://arduinojson.org/v6/api/misc/copyarray/).

Best regards,
Benoit

t0maszduda, 2021-03-26:
Thanks 
I saw copyArray() earlier but I couldn't find example with deserialize 
I don't know how it parse?

StaticJsonDocument<500> doc;

copyArray(values, doc.to<JsonArray>());

float t = bme.readTemperature();
doc["temp"] = t;
float h = bme.readHumidity();
doc["hum"] = h;
float p = bme.readPressure() / 100.0F;
doc["pre"] = p;

serializeJson(doc, s);

////////////////////////////////////////////////////////////////

deserialize

    StaticJsonDocument<300> doc;
    DeserializationError err = deserializeJson(doc, linkSerial);

    if (err == DeserializationError::Ok) 
    {
  
      values????

      t=doc["temp"].as<float>();
      h=doc["hum"].as<float>()
      p=doc["pre"].as<float>();

}

else....

bblanchon, 2021-04-02:
Serialize:

```
copyArray(values, doc["values"]);
```

Deserialize:

```
copyArray(doc["values"], values);
```

t0maszduda, 2021-04-02:
WOW that's easy ;)
Thanks

closed 2021-04-13 by bblanchon