bblanchon/ArduinoJson #1523: createNestedArray

amedeodivito, 2021-03-25:
why the generated json code includes only the first 5 data instead of 6?
who can help me?
`void statoImpianto() {
  doc.clear();
  doc["ident"] = "termo";                                    // identita' dispositivo
  doc["UM"] = Hum;                                           // Umidita'
  doc["TA"] = Tem;                                           // Temperatura attuale
  doc["TC"] = comfort;                                       // Temperatura comfort
  doc["TIME"] = timer;                                       // Tempo attivita' caldaia
  if (stato == 0) doc["SC"] = "Spenta";                      // Stato caldaia
  if (stato == 1 && pausa == true)  doc["SC"] = "Pausa";
  if (stato == 1 && pausa == false) doc["SC"] = "Accesa";

  JsonArray aux_io = doc.createNestedArray("IO");
  aux_io.add(digitalRead(OUT_1));                           // Stato porta 1
  aux_io.add(digitalRead(OUT_2));                           // Stato porta 2
  aux_io.add(digitalRead(OUT_3));                           // Stato porta 3
  aux_io.add(digitalRead(OUT_4));                           // Stato porta 4
  aux_io.add(digitalRead(OUT_5));                           // Stato porta 5
  aux_io.add(digitalRead(OUT_6));                           // Stato porta 6

  // Genera il record JSON e lo salva su variabile output
  char output[BUFLEN];
  serializeJson(doc, output);
  Serial1.println(output);
  delay(50);
}`

bblanchon, 2021-03-26:
Hi @amedeodivito,

Please read [Why is the output incomplete?](https://arduinojson.org/v6/issues/incomplete-output/)

Best regards,
Benoit

amedeodivito, 2021-03-26:
thanks, i solved
sorry for disturbing

closed 2021-03-26 by bblanchon