bblanchon/ArduinoJson #1525: error: a15 cannot be used in asm here

Tony763, 2021-03-27:
Hello,
  I have a problem compiling `deserializeJson` example in PlatformIO and Espressif 8266 (2.6.2).

Code is placed in function called from class function:
```
StaticJsonDocument<200> doc_in;
DeserializationError error = deserializeJson(doc_in, msgch);

if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
}
```
Above code gives me error ` error: a15 cannot be used in asm here`.

Problem will be with use of  `error.f_str()` as if I comment whole `Serial.println(error.f_str());` then code will compile.

bblanchon, 2021-03-27:
Hi @Tony763,

This seems to be a problem in the ESP8266 core, can you try to upgrade/reinstall it?
If this doesn't work, replace `error.f_str()` with `error.c_str()`.
You may also need to set [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`.

Best regards,
Benoit

Tony763, 2021-03-27:
Hi @bblanchon, I updated to latest version of ESP core without change.

`error.c_str()` this works, could it be mentioned in examples?
Thank you.

jjsuwa-sys3175, 2021-03-28:
some information:
- ESP8266 Arduino cores before 8th July 2020 have the issue (see https://github.com/esp8266/Arduino/commit/d979b57d7640b78d59d7d59db744fe75bbcee9ac).

bblanchon, 2021-03-28:
@jjsuwa-sys3175, thanks for the information.
I confirm that I can compile with 2.7.4 without any error.

Tony763, 2021-03-28:
Hi, I use PlatformIO allows me only to update to version 2.6.3. I will open a issue there. Thank You

closed 2021-03-28 by Tony763

bblanchon, 2021-03-29:
@Tony763, there seem to be two versions in the Espressif 8266 platform on PlatformIO: the platform version and the framework version.

[image]

As you can see, I installed 2.6.3, but the framework is 2.7.4.
I find this really confusing, and I'd love to get an explanation...
@jjsuwa-sys3175, do you know why it's like that?

Tony763, 2021-03-29:
Yes I found out that I mistook them. I raised an issue on platformio repository with ask for help.

Tony763, 2021-03-29:
My version is exactly a same.
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.3) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
```
Did you successfully build the example with Arduino IDE and Platform IO or just with one  of them?

bblanchon, 2021-03-29:
I build it successfully with both.
The CI contains a [build with PlatformIO on ESP8266](https://github.com/bblanchon/ArduinoJson/runs/2208544122?check_suite_focus=true#step:5:75) as well.