bblanchon/ArduinoJson #1527: No Known Method to Check If Zero Valued Integer/Boolean Key Exist

NarinLab, 2021-03-31:
Hi, 

I'm working on ESP32 with ArduinoJson 6.17.3.

I have a configuration struct like this:

```
struct ConfigCoMCU
{
  uint8_t flagRly1;
  uint8_t flagRly2;
  uint8_t flagRly3;
  uint8_t flagRly4;

  uint8_t pinRly1;
  uint8_t pinRly2;
  uint8_t pinRly3;
  uint8_t pinRly4;
  uint8_t pRlyOn;
  uint8_t pRlyOff;
};
```

and I also have a setter function that will set the configuration struct from JSON value:
```
void Madava::setConfigParam(StaticJsonDocument<DOCSIZE> &doc)
{
  if(doc["flagRly1"]){
    configMain.flagRly1 = doc["flagRly1"].as<uint8_t>();
  }
  if(doc["flagRly2"]){
    configMain.flagRly2 = doc["flagRly2"].as<uint8_t>();
  }
  if(doc["flagRly3"]){
    configMain.flagRly3 = doc["flagRly3"].as<uint8_t>();
  }
 //and so on...
```
my question is, how can i check if ```doc``` contains a specific key that also have a zero/false value? If i do like above, then if the value of doc["flagRly1"] is 0 for example, then the if statement of the setter will be skipped. In other word, i can only set a value other than 0 for those configuration parameter, i.e 1.

```doc``` might containt one, two, or other key that is not on the struct. So key exist check is really required to determine which part of the struct the key: value pair belongs to.

I'm also trying to use a helper function, converting the value into String in converting again into bool and then make a check using if statement, but if i have more than ten of config parameter this would be so complex to see and to write. Other solution like ```.containsKey``` method requiring me to convert ```doc``` into JsonObject that also i think not so efficient.

Please help, and thankyou very much.

NarinLab, 2021-03-31:
Update, 

i'm sorry for not trying a nullprt comparison. Here is the answer:

```
(doc["fRly1"] != nullptr) ? log_i("fRly1 is not nullptr") : log_i("fRly1 is nullptr");
```

closed 2021-03-31 by NarinLab

bblanchon, 2021-03-31:
Hi @NarinLab,

There are three ways to do that:

* `if (doc["key"] == nullptr)`
* `if (doc["key"].isNull())`
* `if (doc.containsKey("key"))`

Yes; [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) supports [containsKey()](https://arduinojson.org/v6/api/jsondocument/containskey/).  
No; [`JsonDocument::as<JsonObject>()`](https://arduinojson.org/v6/api/jsondocument/as/) is not an expensive operation.

BTW, you can change the parameter type of your function from `StaticJsonDocument<DOCSIZE>&` to `const JsonDocument&`.

Best regards,
Benoit