bblanchon/ArduinoJson #1529: Non-English UTF8 characters cause ESP32 restart.

YordanYanakiev, 2021-04-02:
Hello.
I am having a ASP.NET Web API Server which is proving JSON which contains strings in local ( Bulgarian ) character set.
The initial tests I've performed was with standard English character set, and everything worked flawlessly, but in the moment I've switched to Bulgarian ( Cyrillic ) character set - it turns out that .as<String>() start getting weird additional characters, and in a couple of request it was leading to restarting of ESP32 with "bad HEAP" error. ( probably reading outside the space of the string memory ).

Here a short of what I'm doing:

I have this structure :

```
struct Message
{
    int id;
    int type;
    String caption;
    String text;
};
```
initialized like that :

`#define MAX_MESSAGES_COUNT 32`
`Message messageRequests[ MAX_MESSAGES_COUNT ];`

Then :

```
DynamicJsonDocument doc( 8192 );
    deserializeJson( doc, json );
    JsonObject obj = doc.as<JsonObject>();

    bool isThereDifferenceWithPrevious = false;
    int currentMessagesSize = 0;
    Message currentRequests[ MAX_MESSAGES_COUNT ];
   
    if( doc[ "messages" ].isNull() )
    {
        if( messageRequests[ 0 ].id > -1 )
            isThereDifferenceWithPrevious = true;
    }
    else
    {
        currentMessagesSize = obj[ "messages" ].size();
        if( currentMessagesSize > 0 )
        {

            for( int c = 0; c < currentMessagesSize; c++ )
            {
                JsonObject o = obj[ "messages" ][ c ].as<JsonObject>();

                currentRequests[ c ].id      = o[ "id" ].as<int>();
                currentRequests[ c ].type    = o[ "type" ].as<int>();

                currentRequests[ c ].caption = o[ "caption" ].as<String>();
                currentRequests[ c ].text    = o[ "text" ].as<String>();
                
            }

```

In the last two rows it happens the crash seems like.

closed 2021-04-03 by YordanYanakiev

bblanchon, 2021-04-03:
Hi @YordanYanakiev,

You closed the issue; does it mean you found a solution?
If so, can you tell us a bit about the underlying problem?

Best regards,
Benoit

YordanYanakiev, 2021-04-03:
Sorry. It was related to my code jumping pointer right after the assignment i've pointed above. 
This leads me initially to conclusion that it's an issue inside the json with Cyrillic ( have passed across such in the past ), but it turns out that multithreading made a jumping pointer which I've missed.
Sorry again.
Your code is fine.