bblanchon/ArduinoJson #1530: deserializeJson() failed:  NoMemory. size calculated through the assistant

Z01NE, 2021-04-02:
`#include <iostream>
#include "ArduinoJson.h"

int main() {
    DynamicJsonDocument doc(1536);

    auto file = R"(
        {
            "ssid": "12345678912345678912",
            "pwd": "123456",
            "alarms": [
                {
                "title": "azerrttty",
                "min": 1,
                "hour":2,
                "dw":[0,1,2,3,4,5,6],
                "dm":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20],
                "enabled": false,
                "once": false,
                "id": 1,
                "device": 1,
                "alarmType": 0
                },
                {
                "title": "azerrttty2",
                "min": 1,
                "hour":2,
                "dw":[0,1,2,3,4,5,6],
                "dm":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20],
                "enabled": false,
                "once": false,
                "id": 1,
                "device": 1,
                "alarmType": 0
                }
            ]
        })";

    DeserializationError error = deserializeJson(doc, file);
    if (error)
    {
        printf("deserializeJson() failed: \n");
        printf(error.c_str());
    }
    else
    {
        serializeJsonPretty(doc, std::cout);
    }

  return 0;
}`

bblanchon, 2021-04-03:
Hi @Z01NE,

What processor are you using?
On x86, the assistant says 2048 for example.
Make sure you select the right processor in step 1.

Best regards,
Benoit

Z01NE, 2021-04-03:
Hi @bblanchon,

noob yes wrong architecture.
thank you and great library by the way.

closed 2021-04-03 by Z01NE