bblanchon/ArduinoJson #1531: is it possible to change doc("name") to doc(stringvariable)

pvdw0310, 2021-04-03:
Hi, thanks for the big effort to make that library, I love it. But I'am new to json.

I have a structure like this in my Esp8266 : 

``````
typedef struct 
{ String wifi_id;            // wifimanager variable id
  String wifi_lengte ;     // wifimanager variable invoerlengte
  String wifi_info;          // wifimanager variable info die op website staat
  String wifi_waarde;        // wifimanager variable waarde
  
} wifi_gegevens;

#define max_wifi 5   // maximum aantal variabelen in tabel var
wifi_gegevens wifivar[max_wifi] = 
{
  { "ssid", "20", "SSID", ""},                    // wifi ssid
  { "pasw", "20", "Password", ""},          // wifi password
  { "hostnaam", "20", "Site Passw", ""},    
  { "staticip", "16", "Static Ip Address", "192.168.1.200"},
  { "gateway", "16", "Gateway Address", "192.168.1.1"}
};

DynamicJsonDocument Doc(300);
DeserializationError error = deserializeJson(Doc, line);
  if (error){
      Serial.printf("====> Failed to read file : %s\n", path);
      Serial.println(F("====> Default config ingesteld"));
  }

    wifivar[0].wifi_waarde = Doc["ssid"];    
        
``````
I like to change Doc["ssid"];  with a variable,  so I can use =>  Doc[wifivar[0].wifi_id]);   But I don't find a way to do that.

Is it possible to do that? I don't find any examples.

Many thanks to guide me in the right direction.

bblanchon, 2021-04-04:
Hi @pvdw0310,

`Doc[wifivar[0].wifi_id])` should work, did you try?

Best regards,
Benoit

pvdw0310, 2021-04-04:
Hi Benoit,

I got a compiler error.
But I will have a look, perhaps I made also an other mistake.

 I send you the info later. 
Thanks
Peter

bblanchon, 2021-04-05:
Hi Peter,

Can you copy/paste the compiler output here?

Best regards,
Benoit

pvdw0310, 2021-04-05:
Hi Benoit,

I changed to :

`````` while(file.available()) {
    line = file.readStringUntil('\n');

    DynamicJsonDocument Doc(300);
    DeserializationError error = deserializeJson(Doc, line);
  if (error){
      Serial.printf("====> Failed to read file : %s\n", path);
      Serial.println(F("====> Default config ingesteld"));
  }

    wifivar[0].wifi_waarde = Doc[wifivar[0].wifi_id];       // <==== change and now I get an error
``````

the compiler error :

``````Arduino: 1.8.13 (Windows 7), Board:"NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

..............

In file included from C:\Users\peter\Documents\Arduino\A-productie\wifimanager_configfile_variable_kort\wifimanager_configfile_variable_kort.ino:49:0:

C:\Users\peter\AppData\Local\Temp\arduino_build_79986\sketch\wifimanager.h: In function 'bool readconfigFile()':

wifimanager.h:104:28: error: ambiguous overload for 'operator=' (operand types are 'String' and 'ArduinoJson6152_0000010::enable_if<true, ArduinoJson6152_0000010::MemberProxy<ArduinoJson6152_0000010::JsonDocument&, String> >::type {aka ArduinoJson6152_0000010::MemberProxy<ArduinoJson6152_0000010::JsonDocument&, String>}')

     wifivar[0].wifi_waarde = Doc[wifivar[0].wifi_id];

                            ^

C:\Users\peter\AppData\Local\Temp\arduino_build_79986\sketch\wifimanager.h:104:28: note: candidates are:

In file included from C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/IPAddress.h:24:0,

                 from C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,

                 from C:\Users\peter\Documents\Arduino\A-productie\wifimanager_configfile_variable_kort\wifimanager_configfile_variable_kort.ino:24:

C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/WString.h:95:18: note: String& String::operator=(const String&)

         String & operator =(const String &rhs);

                  ^

C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/WString.h:96:18: note: String& String::operator=(const char*)

         String & operator =(const char *cstr);

                  ^

C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/WString.h:97:18: note: String& String::operator=(const __FlashStringHelper*)

         String & operator = (const __FlashStringHelper *str);

                  ^

C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/WString.h:99:18: note: String& String::operator=(String&&)

         String & operator =(String &&rval);

                  ^

C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/WString.h:100:18: note: String& String::operator=(StringSumHelper&&)

         String & operator =(StringSumHelper &&rval);

                  ^

Bibliotheek ESP8266WiFi op versie 1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\ESP8266WiFi  wordt gebruikt

Bibliotheek ESPAsyncTCP op versie 1.2.0 in map: C:\Users\peter\Documents\Arduino\libraries\ESPAsyncTCP  wordt gebruikt

Bibliotheek ESPAsyncWebServer op versie 1.2.3 in map: C:\Users\peter\Documents\Arduino\libraries\ESPAsyncWebServer  wordt gebruikt

Bibliotheek Hash op versie 1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\Hash  wordt gebruikt

Bibliotheek LittleFS op versie 0.1.0 in map: C:\Users\peter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\LittleFS  wordt gebruikt

Bibliotheek ArduinoJson op versie 6.15.2 in map: C:\Users\peter\Documents\Arduino\libraries\ArduinoJson  wordt gebruikt

Bibliotheek DoubleResetDetector op versie 1.0.2 in map: C:\Users\peter\Documents\Arduino\libraries\DoubleResetDetector  wordt gebruikt

exit status 1

ambiguous overload for 'operator=' (operand types are 'String' and 'ArduinoJson6152_0000010::enable_if<true, ArduinoJson6152_0000010::MemberProxy<ArduinoJson6152_0000010::JsonDocument&, String> >::type {aka ArduinoJson6152_0000010::MemberProxy<ArduinoJson6152_0000010::JsonDocument&, String>}')
``````

pvdw0310, 2021-04-05:
Hi Benoit,

I made some extra changes :
``````
String w;
  for (int i = 0; i < max_wifi; i++){
    w = Doc[wifivar[i].wifi_id];
    wifivar[i].wifi_waarde = w;
  }
``````
that give a compiler error
and if I change it to :
``````

  for (int i = 0; i < max_wifi; i++){
    String w = Doc[wifivar[i].wifi_id];
    wifivar[i].wifi_waarde = w;
  }
``````
then it works perfect.

pvdw0310, 2021-04-05:
Hi Benoit
Back again.  Putting the table to a json string and writing to SPIFFS works perfect
``````
  path = "/config.txt";
  
  DynamicJsonDocument Doc(300);

    for (int i = 0; i < max_wifi; i++){     // vul alle wifimanager var tabel gegevens in config.txt
      Doc[wifivar[i].wifi_id] = wifivar[i].wifi_waarde;
    }

    String str;
    serializeJson(Doc, str);
    Serial.println(str);

    Serial.printf("Writing file: %s\r\n", path);

    File file = LittleFS.open(path, "w");
    if (!file) {
      Serial.println(F("Failed to open config file for writing"));
      return false;
    }

  
    if(file.print(str)){
      Serial.println("- file written");
    } else {
      Serial.println("- write failed");
    }
    file.close();
``````

pvdw0310, 2021-04-07:
Thanks, Don't worry about the delayed response. For some other libraries I have to wait weeks. 

That is a very clear explanation.
Best regards
Peter

closed 2021-04-07 by bblanchon