bblanchon/ArduinoJson #1532: GitHub.ino code with JSON6 book - Http status: -1

fscii, 2021-04-03:
Sorry for noob question but I bought the book JSON 6 in spring 2020, not sure if that means its outdated since I'm only starting on it now.  That being said...

Using provided code example "GitHub.ino" on an ESP8266 d1 mini in Arduino IDE 1.8.13 and ArduinoJSON 6.17.3

I followed book instructions to get GitHub key (pp94-95) and also update certificate thumbprint (pp 97).

Result:  Unexpected HTTP status: -1

I have checked the FAQ and here to see if anyone had mentioned this and found nothing.
Also, as I am nOOb I'm unable to figure out how to fix this.

Please advise if my book/code is simply out of date as well as how to fix this?

MUCH thanks in advance and apologies if I posted this in the wrong place (advise where it should be since I didn't see a discussion thread here).

bblanchon, 2021-04-04:
Hi @fscii,

First, thanks a lot for buying my book, it means a lot to me!

Also, thank you for reporting this issue.
Indeed, I get the same error.

After enabling the log in HTTPClient, I can see:

```
[HTTP-Client][begin] url: https://api.github.com/graphql
[HTTP-Client][begin] host: api.github.com port: 443 url: /graphql
[HTTP-Client][begin] httpsFingerprint: 968407df0b1cf65814dfd7333557519b154d8ce7
[HTTP-Client][sendRequest] type: 'POST' redirCount: 0
[HTTP-Client] connected to api.github.com:443
[HTTP-Client] transport level verify failed
[HTTP-Client][returnError] error(-1): connection failed
``` 

I'm looking for a solution...

Best regards,
Benoit

bblanchon, 2021-04-04:
Switching to the new API fixes the issue!

Instead of:

```c++
http.begin(F(GITHUB_URL), F(GITHUB_FINGERPRINT));
```

you must do:

```c++
WiFiClientSecure client;
client.setFingerprint(GITHUB_FINGERPRINT);
http.begin(client, F(GITHUB_URL));
```

This also removes the deprecation warning.

I'll upload a new `sketchbook6.zip` in a minute.

bblanchon, 2021-04-13:
@fscii, did you try the new version?

fscii, 2021-04-14:
@bblanchon yes I did and it works.  Thank you very much for helping.
I'm going through each example as I get time.  I did notice also SPIFFS deprecated and in the esp01 example using SPIFFS I simply did a "search replace" with LittleFS and it works perfectly.  Onto the Adafruit example, can get it all to work except that map.

Maybe its useful for a noob to go through all of this with noob eyes lol

Thanks again!

bblanchon, 2021-04-15:
Indeed, SPIFFS is deprecated but still works.
I'll update the examples in the next revision of the book.
Let me know if you need anything else!

closed 2021-04-15 by bblanchon