bblanchon/ArduinoJson #1534: Runtime exception when upgrading from 6.17.2 to 6.17.3

sblantipodi, 2021-04-06:
Hi,
thanks for your wonderful work here Benoit.

I have upgraded from 6.17.2 to 6.17.3 and I'm getting a strange runtime exception on this code:
https://github.com/sblantipodi/glow_worm_luciferin

It runs on an ESP8266.

If I revert to 6.17.2 the runtime exception vanish.

It's really weird. Any idea?

Thanks

bblanchon, 2021-04-07:
Hi @sblantipodi,

I can see no change between 6.17.2 and 6.17.3 that could explain a crash.
I suppose that you're running into undefined behavior.

It's hard to tell where the problem is located because the code is quite long.
However, there is something fishy in `callback()`: this function uses the "zero-copy" mode of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/), and stores the result in a global [JsonDocument](https://arduinojson.org/v6/api/jsondocument/), so there are probably some dangling pointers in `bootstrapManager.jsonDoc`.

Please reproduce the crash and use [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder).
Even if this tool is not always reliable, it will give us a better view of the problem.

Best Regards,
[Benoit](https://paypal.me/benoitblanchon)

sblantipodi, 2021-04-07:
@bblanchon thanks for the answer. It seems that I can't reproduce the stack trace anymore
ESP simply reboots when caling  deserializeJson().

This is really weird since with 6.17.2 it works like a charm.

Do you see some problems in this code?

This is the callback for the pubsubclient lib
```
void callback(char *topic, byte *payload, unsigned int length) {

      bootstrapManager.jsonDoc.clear();
      deserializeJson(bootstrapManager.jsonDoc, payload); // THIS IS WHERE ESP REBOOTS
      int numLedFromLuciferin = bootstrapManager.jsonDoc[LED_NUM_PARAM];
...
```

this is the BootstrapManager class:
```
class BootstrapManager {

  public:
    StaticJsonDocument<BUFFER_SIZE> jsonDoc;

 ...
```

as you can see bootstrapManager.jsonDoc is initialized in the BootsrapManager class,
I use that jsonDoc everywhere in my code to access the jsonDoc.

can this create dangling pointers ?

sblantipodi, 2021-04-07:
Ok,
since my code isn't changed and I think that a dangling pointer should not have a repeatable behaviour like I have,
I have made a diff between 6.17.2 and 6.17.3 and discovered that the two libs are very very similar apart the updated 
comment "Copyright Benoit Blanchon 2014-**2021**" that touches all the files.

The part that make my ESP crash and reboot is this:
https://github.com/bblanchon/ArduinoJson/blob/803e07c188d609a3a2f405648f815a68ea31f4e5/src/ArduinoJson/Document/JsonDocument.hpp#L318

If I update to 6.17.3 and then manually remove that single line of code, my code works without problems.

any help would be appreciated :)

bblanchon, 2021-04-08:
@sblantipodi, thanks for narrowing the change to just one line.
To me, this confirms my diagnosis: there is an undefined behavior somewhere.
I don't believe there is a problem with the library itself since it still passes GCC and Clang's sanitizers with 99% of coverage.

Regarding the call to `deserializeJson()`: I recommend that you add a third argument to specify the input size.
This could explain the crash on this line, but 6.17.2 would also be affected, so I don't think it's the reason.

Yes. This call creates dangling pointers because the `JsonDocument` stores pointers to the `payload` buffer, which you don't own and could be discarded after `callback()` returns. You can fix this problem easily by casting `payload` to a `const` pointer (don't forget to increase the capacity of the `JsonDocument` accordingly).
However, I don't think that the reason for the crash because the ESP would reboot later, not in the call to `deserializeJson()`

I don't think these changes will fix the crash; please reduce your program to an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example), so we can figure this out.

sblantipodi, 2021-04-08:
6.17.3 destroyed my firmware, I understand that the problem is not inside the library but in my software,
but pretty everything stopped working correctly...

with 6.17.3 this snippet reboots the ESP

main.cpp
```
  DynamicJsonDocument numLedDoc(1024);
  numLedDoc[LED_NUM_PARAM] = dynamicLedNum;
  bootstrapManager.writeToLittleFS(numLedDoc, LED_NUM_FILENAME);
```

```
void BootstrapManager::writeToLittleFS(DynamicJsonDocument jsonDoc, String filename) {

  File jsonFile = LittleFS.open("/" + filename, "w");
  if (!jsonFile) {
    helper.smartPrintln("Failed to open [" + filename + "] file for writing");
  } else {
      serializeJsonPretty(jsonDoc, Serial);
      serializeJson(jsonDoc, jsonFile);
    jsonFile.close();
    helper.smartPrintln("[" + filename + "] written correctly");
  }
}
```

is there something wrong in this code?

Trying to create an MCVE but even this snippet make my ESP reboot, reverting to 6.17.2 and it works well.

thanks!

jjsuwa-sys3175, 2021-04-08:
it seems `DefaultAllocator::freePool()` is called twice after returning from `bootstrapManager.writeToLittleFS()`; because of implicit copy of `jsonDoc` (call-by-value).

workaround: changing to call-by-reference:
```
- void BootstrapManager::writeToLittleFS(DynamicJsonDocument jsonDoc, String filename) {
+ void BootstrapManager::writeToLittleFS(const DynamicJsonDocument& jsonDoc, const String& filename) {
```

bblanchon, 2021-04-09:
`freePool()` is called twice because there are two `DynamicJsonDocument`s.

I ran the program from your previous message and it produced the expected output:

```
{
  "VAL": 5
}OK
```

No crash, no reboot.

Then I [traced](https://github.com/bblanchon/ArduinoTrace) all the calls to make sure there was no [double free](https://cwe.mitre.org/data/definitions/415.html) as you suspected.
I also tried with a large memory pool to make sure the copy construction fails gracefully.
I even tried move construction instead of copy construction: still no problem.

For all my tests, I used an Adafruit HUZZAH and Arduino Core for ESP866 2.7.4

sblantipodi, 2021-04-09:
Hi, thanks for the patience.
Can it be an "out of memory problem"?

I have a lot of functions in my header file

```
bool processUpdate(StaticJsonDocument<BUFFER_SIZE> json);
bool processJson(StaticJsonDocument<BUFFER_SIZE> json);
bool processFirmwareConfig(StaticJsonDocument<BUFFER_SIZE> json);
bool processGlowWormLuciferinRebootCmnd(StaticJsonDocument<BUFFER_SIZE> json);
bool processUnSubscribeStream(StaticJsonDocument<BUFFER_SIZE> json);
bool swapMqttTopic(StaticJsonDocument<BUFFER_SIZE> json);
... many other function
```

BUFFER_SIZE is pretty big and is set at 200.
Can this consume memory?

reducing that variable to 50 seems to have "solved" the problem.

closed 2021-04-09 by sblantipodi

bblanchon, 2021-04-09:
Passing a `StaticJsonDocument` by value can indeed produce a stack overflow.
Instead, you should pass the `JsonDocument` by const-reference.