bblanchon/ArduinoJson #1536: error 'pgm_read_ptr' argument

Talkwondo, 2021-04-11:
Hello,
I am trying to work with arduino cure due and get this weird error:

there are no arguments to 'pgm_read_ptr' that depend on a template parameter, so a declaration of 'pgm_read_ptr' must be available [-fpermissive]

return reinterpret_cast<T>(pgm_read_ptr(p));
in the gmspace_generic.hpp

also,
error: 'pgm_read_ptr' was not declared in this scope

how to fix it?

thanks,

bblanchon, 2021-04-11:
Hi @Talkwondo,

You can fix this error by defining [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`.

However, this error shouldn't happen in the first place.
This most likely comes from a missing macro in Arduino's headers.
What Arduino Core are you using?

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

Talkwondo, 2021-04-11:
I had a pronblem with the arduino core now i get the famus problem:
reinterpret_cast<T>(pgm_read_ptr(p))
'const void*' is not a pointer-to-object type

bblanchon, 2021-04-11:
> 'const void*' is not a pointer-to-object type

This is https://github.com/arduino/ArduinoCore-API/pull/118

Which Arduino Core are you using? (what hardware if you prefer)

closed 2021-04-27 by bblanchon