bblanchon/ArduinoJson #1537: The best JSON library? And why is it absolutely incompatible to already known standards?

Reneg973, 2021-04-13:
Sorry, I just found the homepage and read a few of the "how to use" examples. I am confused why something like the well known and accepted STL library was not taken into account as model for this interface.
E.g. use stream operators '<<' / '>>' for any serialization/deserialization

Other points:
- Overflow safety => if value overflows, it should return the nearest value, not 0

- > Thread Safety: As long as you use a different JsonDocument in each thread, there is no risk of race condition.  

... mmmmh LOL? Every object is threadsafe as long as it is used only in one thread
So I assume if I use this library on ESP32 dual core, where second core streams into a json, while first core queries some values, this won't work reliably?

Anyway good work, please go on

bblanchon, 2021-04-13:
@Reneg973, is this supposed to be satire? or do you actually expect me to answer?

Reneg973, 2021-04-13:
Excuse me, it's my kind. Don't see it as personal attack!
It would be good if you could tell me about your meaning. Perhaps you agree or disagree (and tell me the cause).

bblanchon, 2021-04-13:
Your message sure sounded like a troll :trollface:

> E.g. use stream operators '<<' / '>>' for any serialization/deserialization

I don't think `<<` and `>>` provide an elegant syntax, but if you really love them, you can write the overloads yourself:

```c++
std::ostream& operator<<(std::ostream& os, const JsonDocument& doc) {
  serializeJson(doc, os);
  return os;
}

std::istream& operator>>(std::istream& is, JsonDocument& doc) {
  DeserializationError error = deserializeJson(doc, is);
  if (error) is.setstate(std::ios_base::failbit);
  return is;
}
```

Also, remember that the STL is not available on all Arduino core.

> Overflow safety => if value overflows, it should return the nearest value, not 0

This is your opinion.  
I decided to return zero to be consistent with the "missing value" behavior.
Either way, it's better than letting the integer overflow.

> Every object is threadsafe as long as it is used only in one thread

This is true only if there is no shared state between the objects.
ArduinoJson behaves correctly because it doesn't use any global variable.
Unfortunately, this is not the case for all Arduino libraries.

Reneg973, 2021-04-13:
I see, but trolls just have an opinion and no idea. As Dirty Harry already told us ;)
> Well, opinions are like assholes. Everybody has one.

Leading to this point:
> Overflow safety => if value overflows, it should return the nearest value, not 0
> > This is your opinion.

Sorry to tell you that this is not an opinion, it's a question of usefulness. A return value of 0 doesn't give any assumption about what could be and therefore its return value has no meaning at all.
For example take the function strcmp(), it returns -1 if first parameter is smaller than second one. This information is enough for an ordered map to know the direction of where to insert the string.
So if I just want to check whether a value is negative, I could write `if(json[".."].as<int8>() < 0)`, which works, even if complete conversion is not required. Ok, how can we proceed, if value is not available:
What do you think about:
```
bool isAvailable;
if(json["..."].as<int8>(isAvailable) == 0 && isAvailable)
```
or using new C++
`if(bool isAvailable; (json["..."].as<int8>(isAvailable) == 0 && isAvailable))`

Ok, next point.
I know how to write a wrapper around your functions ;). There are developers which are new to the Arduino community, but developed for Linux or Windows or even mbedOS - like me!
What I really miss with Arduino is a standardized interface language and syntax. Every library I saw up to now has different wording, different handling, different usage - reinventing the wheel again and again and again. I am annoyed of reading the onethousandandfirst library with different API. And so my intention was to give some hints how you could extend your API to support career changers and increase the acceptance of your library. **Note, I like what you did up to now!** I just would like to see some further additions.
When talking about these, we come to OOP. If there are developers coming from a strict OOP language like Java or C#, they expect something like `JsonDocument doc; doc.deserialize(...);` Is this possible? I didn't read anything about it in your examples.

Reneg973, 2021-04-13:
And, as good will let me give you another tip for an improvement.
With your last breaking changes `.as<>()` you force the user to use const for pointer type, while it is not required for a non pointer type.
As the return value of as is already a rvalue and you  should not allow to directly modify the value (as<>() should be const), (I think you did that already), you can let the compiler add the const if missing.
See  also [cppreference](https://en.cppreference.com/w/cpp/types/add_cv). You can use SFINAE with enable_if<is_pointer_type> for this. But I assume you already know.

bblanchon, 2021-04-14:
If you need to know whether a key is present or not, simply use [containsKey()](https://arduinojson.org/v6/api/jsondocument/containskey/):

```c++
if (doc.containsKey["key"])
```

Alternatively, you can compare with `nullptr`:

```c++
if (doc["key"] == nullptr)
```

If you want to know whether a value is negative, simply compare the [JsonVariant](https://arduinojson.org/v6/api/jsonvariant/) with zero:

```c++
if (doc["key"] < 0)
```

This will work even if the value is out of `long`'s range, since the parser automatically promotes larges values to `double`.

I worked 9 years as a C# developer; from what I remember, `doc.deserialize(...)` is neither a convention nor a recommendation.
On the contrary, it's a bad design decision because it couples the DOM with the parser.
Something like `doc = deserialize(...)` would be better, except for error because we don't use exceptions in ArduinoJson.

@Reneg973, I still think you're just trolling around. If you have any specific questions or requests, I'd be glad to read them; otherwise, please stop wasting my time. Thanks.

closed 2021-04-27 by bblanchon