bblanchon/ArduinoJson #1538: This is not an issue but an help request.

sblantipodi, 2021-04-15:
Hi guys,
is there a forum, a chat, or something similar where to talk the ArduinJSON use and best practices?

I'm sorry if I open an issue but can I have some examples on how to use ArduinoJSON ?

Suppose that I'm using an MQTT lib like PubSubClient and I have this callback.

```
void callback(char *topic, byte *payload, unsigned int length) {
  // Use arduino JSON here.
}
```

Now Inside the callback I would like to call some other functions for this purposes.

1) Pass the payload to a function, deserialize it into a JSON object, return the JSON object of some kind, and use it by values like:
int myint = myjsonObj.myvalue1;
String myString = myjsonObj.myvalue2;

2) Create a JSON object of some kind and pass it to a function to serialize it into SPIFFS or LittleFS.

How can I do it without creating danglings objects? What is the right way to do it?

Is there a forum for this kind of question? Can I ask it here?

condemil, 2021-04-16:
The official website has great documentation about the library.

For PubSubClient there is a How To: https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/

And for working with file system there is an example: https://arduinojson.org/v6/example/config/

bblanchon, 2021-04-16:
Hi @sblantipodi,

This is the right place if you want answers from me and loyal users of the library.
If you want to get opinions from more users, I recommend posting to StackOverflow or to the Arduino forum.
I know that GitHub now offers a Discussions feature, but I'm afraid enabling it will give me twice more work.

You didn't give a specific example, so I'll try to guess what your problem is...  
There are two ways to create dangling pointers from the callback.

## Issue 1

The first is to use a global `JsonDocument` like so:

```c++
StaticJsonDocument<256> doc;

void callback(char* topic, byte* payload, unsigned int length) {
  deserializeJson(doc, payload, length);  // 💀 dangling pointers in doc
}
```

Indeed, because `payload` is a non-const pointer, `deserializeJson()` switches to the zero-copy mode and stores pointers instead of copies. To fix this issue, you simply need to cast `payload` to a pointer-to-const, like so:

```c++
StaticJsonDocument<256> doc;

void callback(char* topic, byte* payload, unsigned int length) {
  // cast payload to pointer-to-const to disable ArduinoJson's zero-copy mode
  deserializeJson(doc, (const byte*)payload, length);  // ✔️ copies in doc
}
```

## Issue 2

The second way to create dangling pointers from `callback()` is to save a pointer in a global variable, like so:

```c++
const char *name = NULL;

void callback(char* topic, byte* payload, unsigned int length) {
  StaticJsonDocument<256> doc;
  deserializeJson(doc, payload, length);
  name = doc["name"]; // 💀 pointer is invalid as soon as function returns
}
```

The solution is to make a copy of the string:

```c++
const char name[32] = "";

void callback(char* topic, byte* payload, unsigned int length) {
  StaticJsonDocument<256> doc;
  deserializeJson(doc, payload, length);
  strlcpy(name, doc["name"] | "default", sizeof(name));  // ✔️ copies in name
}
```

Of course, you can use a `String` if you prefer.

I updated the page [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)  to include these sections.

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

sblantipodi, 2021-04-16:
Thank you so much for your patience and for your awesome work here.

closed 2021-04-16 by sblantipodi