bblanchon/ArduinoJson #1540: PubSubClient publish mutates not related StaticJsonDocument

condemil, 2021-04-16:
I am using ArduinoJson to deserialize received mqtt message and send back serialized mqtt response. As suggested by docs, I am not reusing same StaticJsonDocument and use throw-away objects. I see that serializing and sending mqtt message affects StaticJsonDocument in different function.

Minimal code to reproduce:

```cpp
#include <Arduino.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <WiFiClient.h>

WiFiClient _esp_client;
PubSubClient mqtt(_esp_client);

static const char *WIFI_NAME = "";
static const char *WIFI_PASS = "";
static const char *MQTT_HOST = "";
static const char *MQTT_LOGIN = "";
static const char *MQTT_PASS = "";
static const uint16_t MQTT_PORT = 1883;
static const char *MQTT_TOPIC_STATE = "esp8266/test";
static const char *MQTT_TOPIC_COMMAND = "esp8266/test/set";

void publish() {
    StaticJsonDocument<128> doc2;

    doc2["mode"] = "high";

    char publishBuffer[128];
    serializeJson(doc2, publishBuffer);

    mqtt.publish(MQTT_TOPIC_STATE, publishBuffer, true);
}

void callback(char *topic, byte *payload, unsigned int length) {
    Serial.printf("mqtt: message arrived [%s]\n", topic);

    if (length >= MQTT_MAX_PACKET_SIZE) {
        Serial.println(F("mqtt: payload is too long"));
        return;
    }

    StaticJsonDocument<128> doc;
    DeserializationError error = deserializeJson(doc, payload);

    if (error) {
        Serial.println(F("parse_document: failed to deserialize json"));
        return;
    }

    if (doc["state"].is<char *>()) {
        const char* state = doc["state"];
        Serial.printf("parse_document: state is in the document 1: %s\n", state);
    } else {
        Serial.println(F("parse_document: state is not in the document 1"));
    }

    publish(); // if this removed, document starts to return state as expected

    if (doc["state"].is<char *>()) {
        const char* state = doc["state"];
        Serial.printf("parse_document: state is in the document 2: %s\n", state);
    } else {
        Serial.println(F("parse_document: state is not in the document 2"));
    }
}

void setup() {
    Serial.begin(115200);

    WiFi.begin(WIFI_NAME, WIFI_PASS);

    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println(F("wifi: connection failed"));
    }

    Serial.println(F("wifi: connected"));

    mqtt.setServer(MQTT_HOST, MQTT_PORT);

    if (mqtt.connect(MQTT_HOST, MQTT_LOGIN, MQTT_PASS)) {
        Serial.println(F("mqtt: connected"));
        mqtt.subscribe(MQTT_TOPIC_COMMAND);
    } else {
        Serial.printf("mqtt: connect failed, rc=%d\n", mqtt.state());
    }

    mqtt.setCallback(callback);
}

void loop() {
    mqtt.loop();
}
```

MQTT message I send to `esp8266/test/set`:

```
{"state": "ON"}
```

Expected output:

```
...
parse_document: state is in the document 1: ON
parse_document: state is in the document 2: ON
```

Output received:

```
...
parse_document: state is in the document 1: ON
parse_document: state is not in the document 2
```

Hardware / versions:

* ESP8266 nodemcu
* platformio
* framework-arduinoespressif8266 3.20704.0 (2.7.4)
* ArduinoJson@6.17.3
* PubSubClient@2.8

bblanchon, 2021-04-17:
Hi @condemil,

I'm doing a lot of support for PubSubClient lately 😄 

I just looked at the source code of PubSubClient and saw that the same buffer is used for transmitting and receiving.
So, when you call `publish()` inside `callback()`, it mutates the `payload()`.

If you want your [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) to be immune to the changes of `payload`, you need to disable the zero-copy mode of [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) by passing a pointer-to-cont as input:

```c++
void callback(char* topic, byte* payload, unsigned int length) {
  StaticJsonDocument<128> doc;

  // cast payload to pointer-to-const to disable ArduinoJson's zero-copy mode
  DeserializationError error = deserializeJson(doc, (const byte*)payload, length);
  
  // ...
}
```

See also:

* [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

condemil, 2021-04-24:
Thank you for clearing it out. I wouldn't be able to figure out that PubSubClient updates buffer that is used by zero-copy JsonDocument without your help as I was thinking in completely different direction :)

closed 2021-04-26 by bblanchon