bblanchon/ArduinoJson #1541: JSON 6 Book example - Reddit.ino - NoMemory error()

fscii, 2021-04-16:
Using ESP8266, I get the following output:

`Connecting to WLAN...
Connected to WLAN
Connecting to server...
Connected to server
Sending request...
Receive response...
90	Classic snake game on Arduino but pvp with 4 players.
50	Finally got Neopixel working in ESP32-S2. More in discription
23	I made a working miniature attraction ride! (Pt. 1)
685	I made a thermometer-clock without screen, needles or any visible interface.
deserializeJson() failed with code NoMemory
`

Code says:  Tested with ESP8266 core 2.6.3
Except I'm using ESP8266 core 2.7.4
Arduino IDE: 1.8.13
ArduinoJSON library: 6.17.3

Changes I made to code:
The usual SSID/PSK.

Any help would be appreciated.  I'm very interested in seeing how to actually fix the issue to learn how to handle it in the future in my own code.

Thank you in advance.

bblanchon, 2021-04-17:
Hi @fscii,

Thank you very much for purchasing my book 🙏 

I cannot check right now, but it's likely due to a larger response from Reddit's server.
You could fix the issue by increasing the size of the `JsonDocument`, but it's already pretty big (16KB).
A better solution would be to use a filter, like so:

```c++
StaticJsonDocument<48> filter;
filter["data"]["title"] = true;
filter["data"]["ups"] = true;
...
deserializeJson(doc, client, DeserializationOption::Filter(filter));
```

I'll revise the example for the next edition of the book.

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

fscii, 2021-04-17:
"score" instead of "ups" but it works thank you very much!

bblanchon, 2021-04-17:
You're welcome, @fscii.
Good luck with your project!

closed 2021-04-17 by bblanchon