bblanchon/ArduinoJson #1543: Error in Writer.hpp

cccadel, 2021-04-20:
I'm trying to convert my object with a to_json(char out[128]) method. This is what it looks like:

`void Settings::to_json(char out[JSON_OBJECT_MAX_CAPACITY]) {
    StaticJsonDocument<JSON_OBJECT_MAX_CAPACITY> doc;

    doc[JSON_NAME_SERIAL_NUMBER] = m_serial_number;
    doc[JSON_NAME_MEASURE_INTERVAL] = m_measure_interval;

    serializeJson(doc, out);
}`

But I am receiving the following error: `request for member 'write' in '*((ArduinoJson6173_91::Writer<char*, void>*)this)->ArduinoJson6173_91::Writer<char*, void>::_dest', which is of non-class type 'char*'`. This is due to using the out parameter. When the method looks like this:

`void Settings::to_json() {
    StaticJsonDocument<JSON_OBJECT_MAX_CAPACITY> doc;

    doc[JSON_NAME_SERIAL_NUMBER] = m_serial_number;
    doc[JSON_NAME_MEASURE_INTERVAL] = m_measure_interval;

    char out[JSON_OBJECT_MAX_CAPACITY]

    serializeJson(doc, out);
}`

It does not throw this errror, but I like to use the method as I stated above.

bblanchon, 2021-04-21:
Hi @cccadel,

The C programming language treats array parameters as pointers.
In your program, the function signature:

```c++
void Settings::to_json(char out[JSON_OBJECT_MAX_CAPACITY])
```

is translated to:

```c++
void Settings::to_json(char *out)
```

Indeed, the size of the array is irrelevant and is completely ignored.

So, what does that mean for ArduinoJson?
Well, instead of calling the overload taking an array:

```c++
size_t serializeJson(const JsonDocument& doc, char output[size]);
```

your program tries to call an overload that takes a pointer:

```c++
size_t serializeJson(const JsonDocument& doc, char *output);
```

But this overload doesn't exist, so the generic behavior using a [custom writer](https://arduinojson.org/news/2019/11/01/version-6-13-0/) is compiled, and that's where the error comes from.

How to solve it?
Fortunately, [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) has an overload that supports a pointer, but as you can see it requires a third parameter that specifies the size of the buffer:

```c++
size_t serializeJson(const JsonDocument& doc, char* output, size_t outputSize);
```

In other words, you can fix your program by replacing the invocation with:

```c++
serializeJson(doc, out, JSON_OBJECT_MAX_CAPACITY);
```

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

cccadel, 2021-04-21:
Thanks for the answer @bblanchon! That solved it.

closed 2021-04-22 by bblanchon