bblanchon/ArduinoJson #1544: Create an Array of Array from an Array of Object working like an Array

Cabu, 2021-04-24:
Hello, sorry to ask such a question but I try to create a JSON document who should be an array of array but only manage to have a partial result.

```c
const int numPalettes = 16;
CRGBPalette16 g_palettes[numPalettes] = { RainbowColors_p, RainbowStripeColors_p, ..., myBlackPalette_p };

void handleGetPalettes() {
  DynamicJsonDocument doc(512);

  JsonArray data = doc.createNestedArray("PALETTES");
  for (int i = 0; i < numPalettes; i++) {
    JsonArray data2 = data.createNestedArray();
    for (int j = 0; j < 16; j++) {
      data2.add(g_palettes[i][j].r);
      data2.add(g_palettes[i][j].g);
      data2.add(g_palettes[i][j].b);
    }
  }

  String buf;
  serializeJson(doc, buf);
  g_server.send(200, F("application/json"), buf);
}
```

CRGBPalette16  is a class that could operate like an array of struct {byte r; byte g; byte b;}.

I only get the first palette of my array of palettes:
```json
{"PALETTES":[[255,0,0,213,42,0,171,85,0,171,127,0,171,171,0,86,213,0,0,255,0,0,213,42,0,171,85,0,86,170]]}
```

in place of something like 
```json
{"PALETTES":[[255,0,0,...],[...2nd palette...],[...],...]}
```

I must say that my C/C++ fu is heavily rusty (last time I programmed in C was 30 years ago in school) and I just started programming on a D1 Mini 3 weeks ago for the first time :)

bblanchon, 2021-04-25:
Hi @Cabu,

If I'm not mistaken, the final document should look like:

```json
{"PALETTES":[[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47],[0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47]]}
```

This is a pretty big document!
The [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) recommends using a 16KB buffer to store it; that's why the document cannot fit in a buffer of 512B.

Honestly, I don't think you should use JSON to transmit or store a color palette; instead, you should use a binary format.
For example, you could simply do:

```c++
g_server.send(200, F("application/octet-stream"), g_palettes, sizeof(g_palettes));
```

Or, if you want something more "standard", try to implement a palette file format, such as the old .PAL format.

See also:

* [Why is the output incomplete?](https://arduinojson.org/v6/issues/incomplete-output/)
* [Serialization and JSON on Arduino (video)](https://youtu.be/hAB4TdX8dwM)

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

Cabu, 2021-04-25:
Well, I though that sending a JSON document who at max will be of 3200 (16*16*3*4+some) bytes length would be feasable. I should change my mind of how I should do it. But nontheless, I still don't understand why I don't have the 16 arrays or an out of memory error with that code :/ Is it because I reuse data2, Shoudl I use an array of data2 ?

bblanchon, 2021-04-27:
Hi @Cabu,

There is a significant overhead for storing a byte in a `JsonVariant`: on ESP8266, each variant takes 16 bytes in the `JsonDocument`.

ArduinoJson doesn't throw exceptions; however, it returns an error status that tells if the insertion succeed.
Please see:

* [JsonVariant::add()](https://arduinojson.org/v6/api/jsonarray/add/)
* [JsonDocument::overflowed()](https://arduinojson.org/v6/api/jsondocument/overflowed/)

Your original code was correct, as you can see on this demo:
https://wandbox.org/permlink/LDAv9FKVMjsU6RWG

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

Cabu, 2021-04-27:
Thanks a lot !!

closed 2021-04-27 by Cabu