bblanchon/ArduinoJson #1547: Data gone after deserializeJson(); on ESP32 Arduino.

zekageri, 2021-04-29:
### Hi! Thank you for the library! Really really amazing.

**My problem is that i want to use the esp32's external RAM ( 4mb ) to store a large JSON object inside it.**

_Here is my attempt:_

```C
class SHUT {
 private:
      // Create a good looking char variable inside a class.
      char * smartAdjustData = (char *) ps_malloc (20000 * sizeof (char));
 public:
      SHUT(const JsonObject& shut){
         serializeJson(shut["GRAPH"],smartAdjustData,20000); // Serialize a JSON object inside it.
         Serial.println(smartAdjustData) 
         // Okay, this one is printing out well, all my data is inside the smartAdjustData variable.
       }
/* Later in a function i want to work with that data at runtime */

inline void smartAdjust(){
            if(isSmartAdjust && isTuneEnd && graphState){
                if(Time.Get_Min() != lastSmartAdjust_Min){
                    lastSmartAdjust_Min = Time.Get_Min();
                    SpiRamJsonDocument doc(20000);
                    Serial.println(smartAdjustData);
                    DeserializationError error = deserializeJson(doc, smartAdjustData,20000);

// Okay , the first deserialization is working well. I get the JSON data, but
// as you can see i run that function every minute, in a loop and i want to periodically check that data but at 
// second iteration my smartAdjustData is gone. Whats left is just 3 char inside.
// and after the first iteration i just get invalid inputs from the deserialization.

                    if (error) {
                        String Error = "JSON SHUT GRAPH Error ~ ";Error += String(error.c_str());
                        errorLog(Error);
                    }else {
                        Serial.println("Graph ok");
                        JsonObject documentRoot = doc.as<JsonObject>();
                        boolean IsFound = false;
                        for (JsonPair keyValue : documentRoot) {
                            if(Time.Get_Day() == keyValue.key().c_str()){
                                JsonArray Day = documentRoot[Time.Get_Day()];
                                for(int i = 0; i < Day.size(); i++){
                                    if(isTimeReached(Day[i]["x"],Day[i+1]["x"])){
                                        IsFound = true;
                                        uint16_t Set_Percent = Day[i]["y"];
                                        if(Set_Percent != lastPercent){
                                            goToWebPercent(Set_Percent);
                                        }
                                        break;
                                    }
                                }
                                if(IsFound){break;}
                            }
                        }
                    }
                }
            }
        }
 void loop(){
smartAdjust();
}
}
```

**Data inside the variable before the first deserialization:**

`{"mon":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"16:20","y":67},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}],"tue":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"16:20","y":67},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}],"wed":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"16:20","y":67},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}],"thu":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"15:00","y":96},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}],"fri":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"16:20","y":67},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}],"sat":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"16:20","y":67},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}],"sun":[{"x":"00:00","y":44},{"x":"01:00","y":58},{"x":"02:00","y":54},{"x":"03:00","y":47},{"x":"04:00","y":41},{"x":"05:00","y":61},{"x":"06:00","y":54},{"x":"07:00","y":47},{"x":"08:00","y":60},{"x":"09:00","y":60},{"x":"10:00","y":42},{"x":"10:30","y":59},{"x":"11:10","y":47},{"x":"11:50","y":59},{"x":"12:10","y":37},{"x":"12:20","y":65},{"x":"13:10","y":38},{"x":"13:50","y":73},{"x":"14:20","y":41},{"x":"16:20","y":67},{"x":"18:10","y":43},{"x":"19:00","y":65},{"x":"20:30","y":40},{"x":"21:30","y":68},{"x":"22:30","y":60},{"x":"23:30","y":39}]}`

**Data inside the variable after the first deserialization:**

`mon`

bblanchon, 2021-04-30:
Hi @zekageri,

Please see: [Why is the input modified?](https://arduinojson.org/v6/issues/altered-input/)

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

zekageri, 2021-05-01:
Oh, thank you! 
I will try it and report back

zekageri, 2021-05-04:
Thanks for the info and sorry for the issue ticket. That solved so many of my problems, not just that. I have some really big json strings and now i can place all of them into external ram. Thank you again kind sir.

closed 2021-05-04 by zekageri

bblanchon, 2021-05-04:
You're welcome!
Thank you for using ArduinoJson.
Good luck with your project!