bblanchon/ArduinoJson #1548: Issues with Iterators/ Templates when used in Sloeber

OOHehir, 2021-04-29:
Hello,
Many thanks for all your work on this project, its impressive both in terms of it's documentation & usefullness.

My issue may perhaps out of the scope of this project but I wonder if you have any input. I'm using v6.17.3 with the sloeber plugin in Eclipse for an 8266. I'm using it with a class method something like:

`
#include <ArduinoJson.h>
#include <ArduinoJson.hpp>

int className::parseReceivedString() {

	DynamicJsonDocument doc(1024);
	bool updateOutputs = false;
	bool updateEEPROM = false;

	DeserializationError error = deserializeJson(doc, (const char *) bufferArrayGlobal);

	JsonArray array = doc.as<JsonArray>();

	unsigned int i =  array.size();

	// using C++11 syntax (preferred):
	for (JsonPair kv : array) {
	    Serial.println(kv.key().c_str());
	    Serial.println(kv.value().as<char*>());
	}
`
However I'm getting errors associated with size & key like:

/opt/eclipse/arduinoPlugin/libraries/ArduinoJson/6.17.3/src/ArduinoJson/Variant/VariantAs.hpp:94:27: error: no matching function for call to 'variantAs(ArduinoJson6173_91::VariantData*&)'
/opt/eclipse/arduinoPlugin/libraries/ArduinoJson/6.17.3/src/ArduinoJson/Variant/VariantAs.hpp:58:1: error: no type named 'type' in 'struct ArduinoJson6173_91::enable_if<false, ArduinoJson6173_91::Pair>'
/opt/eclipse/arduinoPlugin/libraries/ArduinoJson/6.17.3/src/ArduinoJson/Variant/VariantAs.hpp:64:55: error: no type named 'type' in 'struct ArduinoJson6173_91::enable_if<false, ArduinoJson6173_91::Pair>'
/opt/eclipse/arduinoPlugin/libraries/ArduinoJson/6.17.3/src/ArduinoJson/Variant/VariantAs.hpp:70:61: error: no type named 'type' in 'struct ArduinoJson6173_91::enable_if<false, ArduinoJson6173_91::Pair>'

It looks like it has something to do with stl. Do you happen to know a solution?

Many thanks!

bblanchon, 2021-04-30:
Hi @OOHehir,

[JsonObject](https://arduinojson.org/v6/api/jsonobject/) points to a list of [JsonPair](https://arduinojson.org/v6/api/jsonobject/begin_end/), but not [JsonArray](https://arduinojson.org/v6/api/jsonarray/)!

You must replace your loop with either:

```c++
for (JsonPair kv : doc.as<JsonObject>()) {
  Serial.println(kv.key().c_str());
  Serial.println(kv.value().as<char*>());
}
```

or

```c++
for (JsonVariant element : doc.as<JsonArray>()) {
  Serial.println(element.as<char*>());
}
```

The documentation is not so good after all... any idea how I could improve it?

See also:

* [JsonArray::begin() / JsonArray::end()](https://arduinojson.org/v6/api/jsonarray/begin_end/)
* [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

OOHehir, 2021-05-07:
Benoit,

Apologies for the late reply. Many thanks for the help.

With regards to the documentation I don't think I've seen any non-commercial library that compares in terms of documentation so I think you're doing well! Compared to your documentation stuff like the Linus kernel is incomprehensible!

Thanks again for a great library!

Owen

bblanchon, 2021-05-08:
You're welcome!
Thanks for the kind words, @OOHehir.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: