bblanchon/ArduinoJson #1549: ESp32 Crash after deserializeJson

mycarevault, 2021-04-30:
Looking for little help: I am using ArduinoJson for deserializing JSON string that I get from Bluetooth serial comm. The esp32 board crashes at strcpy or strlcpy.  Kindly throw in some examples as I am still in the learning phase.

Code:
#include <Arduino.h>
#include <ArduinoJson.h>
#include <BluetoothSerial.h>
BluetoothSerial SerialBT;

void receiveBTData(String input);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  SerialBT.begin("luminaAPx");
  Serial.println("The device started, now you can pair it with bluetooth!");
  delay(1000);
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available())
  {
     String incomingChar = SerialBT.readString();
     Serial.println(incomingChar);
     receiveBTData(incomingChar);
  }
}

void receiveBTData(String input){
  StaticJsonDocument<250> docBT;
  DeserializationError err = deserializeJson(docBT,input);
  if (err) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(err.f_str());
    return;
  }
  char commTYP[15];
  //strcpy(commTYP,docBT["comm"]);
  strlcpy(commTYP, docBT["comm"], sizeof(docBT));
  //if(String(commTYP) == "action")
//  {
//    char comMsg[20];
//    strcpy(comMsg,docBT["commsg"]);
//    Serial.println("Succesfully updated ssid: "+String(commTYP)+" password: "+String(comMsg));
//  }
}

Crash:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c595  PS      : 0x00060f30  A0      : 0x800d22bc  A1      : 0x3ffc85e0  
A2      : 0x3ffc8639  A3      : 0x00000000  A4      : 0x0000011b  A5      : 0x3ffc86c0  
A6      : 0x00000001  A7      : 0x3ffc68e0  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x3ffc86b4  A11     : 0x00000085  A12     : 0x3ffc86c0  A13     : 0x00000000  
A14     : 0x00000022  A15     : 0x00000022  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4000c595:0x3ffc85e0 0x400d22b9:0x3ffc8600 0x400d239a:0x3ffc87e0 0x400d3659:0x3ffc8820 0x4009052d:0x3ffc8840

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
The device started, now you can pair it with bluetooth!

bblanchon, 2021-05-01:
Hi @mycarevault,

It looks like `docBT["comm"]` returns `NULL`; you should add a default value.
Please see:

* [Why does my device crash or reboot?](https://arduinojson.org/v6/issues/crash-or-reboot/)

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

mycarevault, 2021-05-01:
Benoit, thanks for your reply, 
Adding a default value is for expecting a null, but whereas in my case, I already have data in the "comm" field. 

//String incomingChar ="{\"comm\" : \"action\", \"msg\" : \"gcnfall\"}";

The issue is if I am declaring a string like the one above, then the deserialize JSON will work like charm, whereas if I pass the same string through bluetooth serial comm and collect the same string as below, I can able print the string which I received, but I get null when I do deserialize JSON
 
then I get to see the same string but deserialize JSON returns nill.  strcpy or strlcpy will crash as eventually there is null coming out.

Best Regards
Sathish

bblanchon, 2021-05-03:
Hi Sathish,

It seems that you are receiving a different string via Bluetooh.
Can you check?

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

mycarevault, 2021-05-04:
Benoit, checked. i get a string like "{\"comm\" : \"action\", \"msg\" : \"gcnfall\"}"

bblanchon, 2021-05-06:
If you're absolutely sure that the string is exactly the same, then we can move Bluetooth out the way.
Please write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) that doesn't involve Bluetooth.

mycarevault, 2021-05-07:
Finally solved the issue. There was space that was interfering with the deserialization. After I remove the space it worked. 
The string that was coming from the BT Serial Comm was "{"comm" : **"action",  "msg"** : "gcnfall"}" and changed to "{"comm" : **"action", "msg"** : "gcnfall"}"

closed 2021-05-07 by mycarevault

bblanchon, 2021-05-07:
The number of spaces between values doesn't matter, both documents should work equally.
The cause of your problem was probably somewhere else.

Anyway, I'm glad you fixed your issue.
Thanks for the update, @mycarevault.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: