bblanchon/ArduinoJson #1551: Compatibility nrf52840

sergioxdev, 2021-05-02:
Hi Guys

someone has already tested compatibility with the nrf52840, or the nrf52x family

Thanks in advance for any replies

bblanchon, 2021-05-03:
Hi @sergioxdev,

I never tried, but I would be very surprised if ArduinoJson were not compatible with this microcontroller.
If you have any problems, please open an issue.

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

sergioxdev, 2021-05-03:
Hi @bblanchon 

just tested I confirm that it works perfectly,
congratulations for the excellent work you have done. 👍 

Sergio

closed 2021-05-03 by sergioxdev

bblanchon, 2021-05-04:
Thank you for the kind words, Sergio.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: