bblanchon/ArduinoJson #1552: concern with python 3

Pages44, 2021-05-04:
Hello, I use Python3 and  I try to send {"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]} to the arduino with Serial.
I receive this "String" on the arduino: msg=b'"{\\"sensor\\":\\"gps\\",\\"time\\":1351824120,\\"data\\":[48.756080,2.302038]}"'
But on the website https://arduinojson.org/, we can read:
"char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

DynamicJsonDocument doc(1024);
deserializeJson(doc, json);

const char* sensor = doc["sensor"];
long time          = doc["time"];
double latitude    = doc["data"][0];
double longitude   = doc["data"][1];"

So I have a concern to send time from arduino to python.
Can you help me?
Did you try with pyhton3?
Thank you in advance.

bblanchon, 2021-05-04:
Hi @Pages44,

I'm sorry, I don't understand your question. Can you clarify?
Can you provide an [MVCE](https://stackoverflow.com/help/minimal-reproducible-example)?

Best regards,
Benoit

Pages44, 2021-05-04:
Hello,

Thanks for your answer.
Here is the Python program:
#!/usr/bin/env python
# -*- coding: utf-8 -*-
# lsusb to check device name
#dmesg | grep "tty" to find port name

import serial,time
import json

if __name__ == '__main__':
    print('Running. Press CTRL-C to exit.')
    with serial.Serial("/dev/ttyACM0",baudrate=115200,timeout=1) as arduino:
        time.sleep(0.1) #wait for serial to open
        if arduino.isOpen():
            print("{} connected!".format(arduino.port))
            try:
                while True:
                    cmd=input("data: ")
                    cmd_json=json.dumps(cmd)
                    print(cmd_json)
                    # arduino.write(cmd_json.encode())
                    #time.sleep(0.1) #wait for arduino to answer
                    
                    while arduino.inWaiting()==0: pass
                    if  arduino.inWaiting()>0: 
                        answer=arduino.readline()
                        answer_str=answer.decode('ascii')
                        print("---> {}".format(answer))
                        # dataList=answer.split("x")
                        # print("led state : {}".format(dataList[0]))
                        # print("Analog input A0 : {}".format(dataList[1]))
                        # print("Analog input A1: {}".format(dataList[2]))
                        
                        arduino.flushInput() #remove data after reading
                            
            except KeyboardInterrupt:
                print("KeyboardInterrupt has been caught.")

Here is the Arduino Program:
#include <ArduinoJson.h>

const int ledPin=13;
String nom = "Arduino";
String msg;
int led7 = 7; // NÂ° de la broche ou est branchÃ©e la LED
int led8 = 8; // NÂ° de la broche ou est branchÃ©e la LED

void setup() 
{
  // put your setup code here, to run once:
Serial.begin(115200);
pinMode(led7, OUTPUT); // Configure la broche 8 de la carte arduino en sortie
pinMode(led8, OUTPUT); // Configure la broche 8 de la carte arduino en sortie
digitalWrite(led7,LOW); 
digitalWrite(led8,LOW); 

}

void loop()
{
  readSerialPort();

  // Allocate the JSON document
  //
  // Inside the brackets, 200 is the capacity of the memory pool in bytes.
  // Don't forget to change this value to match your JSON document.
  // Use arduinojson.org/v6/assistant to compute the capacity.
  StaticJsonDocument<500> doc;

  // StaticJsonDocument<N> allocates memory on the stack, it can be
  // replaced by DynamicJsonDocument which allocates in the heap.
  //
  // DynamicJsonDocument doc(200);

  // JSON input string.
  //
  // Using a char[], as shown here, enables the "zero-copy" mode. This mode uses
  // the minimal amount of memory because the JsonDocument stores pointers to
  // the input buffer.
  // If you use another type of input, ArduinoJson must copy the strings from
  // the input to the JsonDocument, so you need to increase the capacity of the
  // JsonDocument.
  
  
  // Length (with one extra character for the null terminator)
  int msg_len = msg.length() + 1; 

  // Prepare the character array (the buffer) 
  char json [msg_len];

  // Copy it over 
  msg.toCharArray(json, msg_len);
  
    // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, json);

  // Test if parsing succeeds.
  /*if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  */
  
   
  // Fetch values.
  //
  // Most of the time, you can rely on the implicit casts.
  // In other case, you can do doc["time"].as<long>();
  const char* sensor = doc["sensor"];
  long time = doc["time"];
  double latitude = doc["data"][0];
  double longitude = doc["data"][1];
  
  // Print values.
  char time_char[64];
  sprintf (time_char, "%20d" ,time);
  Serial.print(json);
  /*
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
  */
    
}

void readSerialPort()
{
msg="";
if(Serial.available())
{
delay(10);
while (Serial.available() > 0)
{
msg += (char)Serial.read();
}
Serial.flush();
}
}

void sendData() 
{
//write data ledState x sensor1 x sensor2
Serial.print(digitalRead(ledPin));
Serial.print("x");
Serial.print(analogRead(A0));
Serial.print("x");
Serial.print(analogRead(A1));
}

I tested with: data= {"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}
It woks for sensor in python 2 but It doesn't work en Python3.

Best regards.
James44

Pages44, 2021-05-04:
Je renvoie le fichier python avec une modification importante: ligne 20, j'enoie la trame vers l'arduino:

#!/usr/bin/env python
# -*- coding: utf-8 -*-
# lsusb to check device name
#dmesg | grep "tty" to find port name

import serial,time
import json

if __name__ == '__main__':
    print('Running. Press CTRL-C to exit.')
    with serial.Serial("/dev/ttyACM0",baudrate=115200,timeout=1) as arduino:
        time.sleep(0.1) #wait for serial to open
        if arduino.isOpen():
            print("{} connected!".format(arduino.port))
            try:
                while True:
                    cmd=input("data: ")
                    cmd_json=json.dumps(cmd)
                    print(cmd_json)
                    arduino.write(cmd_json.encode())
                    #time.sleep(0.1) #wait for arduino to answer
                    
                    while arduino.inWaiting()==0: pass
                    if  arduino.inWaiting()>0: 
                        answer=arduino.readline()
                        answer_str=answer.decode('ascii')
                        print("---> {}".format(answer))
                                             
                        arduino.flushInput() #remove data after reading
                            
            except KeyboardInterrupt:
                print("KeyboardInterrupt has been caught.")

Je n'obtiens pas le même résultat en python 2 et en python 3. Ainsi, je ne pas pas utiliser tel quel les programme proposé sur le site arduinojson.

Merci d'avance pour vos conseils.

bblanchon, 2021-05-06:
Hi @Pages44,

The difference between Python 2 and Python 3 that could make a difference is the string representation.
By default, Python 2 uses ASCII strings, whereas Python 3 uses Unicode strings.
So you may be missing a `.encode('utf-8')` or a `.decode('utf-8')` somewhere, but it's not obvious by looking at the snippet above.

Make sure that the Arduino receives and transmits the expected messages by printing them to another serial port.

@Pages44, this is clearly not a question about ArduinoJson but about Python or whatever Python implementation you're using, so if my answer doesn't help, please ask Stack Overflow or some other help site.
Thank you for your understanding.

Best regards,
Benoit

Pages44, 2021-05-06:
Hello,

thanks a lot pour the answer. It works now with your indication and other information found on Stack Overflow.

Best regards

Philippe

----- Mail original -----

bblanchon, 2021-05-07:
Awesome! Thanks for the update, Philitppe.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2021-05-07 by bblanchon