bblanchon/ArduinoJson #1553: [Question]: JSON float or int for JSON Document

hasenradball, 2021-05-04:
Hello is it relevant, wenn sending an Json document like that:

[image]

if the Temp, Hum, Pressure values are placed as int or float values?
Benefit in case of speed when deliver via network?

bblanchon, 2021-05-06:
Hi @hasenradball,

Why would you use an integer when each value has a decimal part?

Best regards,
Benoit

hasenradball, 2021-05-06:
Hi Benoit,

thanks for reply.
Let me say 
if I have a signifikant benefit in send and receive response time then I would use an int.
If there is no difference I would use a float.

bblanchon, 2021-05-07:
If you send via a good old serial connection, the transmission time is proportional to the number of bytes in the message.
From what I see, it could save 9% in the transmission time.

If you use another medium, like WiFi or Ethernet, the duration will be dominated by the number of packets sent.
For these 150 bytes, the difference would be neglectable.

Keep in mind that JSON is not a performance-oriented format.
If the size is a big concern, use a binary format.
In your case, a C struct would do the job nicely.

hasenradball, 2021-05-07:
Hi Benoit,

thanks for clarification and thanks for the hint:

In the past I handled some point with a structure (example: save data as bin file).
For doing that I placed the data of the structure in and byte array like that:

Lets say having a structure like that:
```
struct Messung {
  time_t zeit;
  float Messwert1;
  float Messwert2;
  float Messwert3;
  float Messwert4;
} DataStructure;
```
I do reconvert the data as:
```
void Sensor_BME280::save_Measurement_toFile(struct Messung* buffer) {
  DBG__PRINT(">>> Schreibe Messung in Datei");
  File fh;
  byte anz;
  time_t zeit;
  short wert1, wert2, wert3, wert4;
  byte help_arr[12];

  // hole Daten aus dem Buffer
  zeit = buffer->zeit;
  wert1 = short(buffer->Messwert1 * 100 + 0.5);
  wert2 = short(buffer->Messwert2 * 100 + 0.5);
  wert3 = short(buffer->Messwert3 * 100 + 0.5);
  wert4 = short(buffer->Messwert4 * 10 + 0.5);

  help_arr[0] = byte(zeit);
  help_arr[1] = byte(zeit >> 8);
  help_arr[2] = byte(zeit >> 16);
  help_arr[3] = byte(zeit >> 24);

  help_arr[4] = byte(wert1);
  help_arr[5] = byte(wert1 >> 8);

  help_arr[6] = byte(wert2);
  help_arr[7] = byte(wert2 >> 8);

  help_arr[8] = byte(wert3);
  help_arr[9] = byte(wert3 >> 8);

  help_arr[10] = byte(wert4);
  help_arr[11] = byte(wert4 >> 8);

  fh = SPIFFS.open(_filename, "a");
  anz = fh.write(help_arr, 12);
  fh.close();
  DBG__PRINT("\t", anz, " Bytes geschrieben\n\n");
}
```

With your answer a Question is comming up...
... do I need reconverting the data ?
or is it also possible to do something like this:
```
anz = fh.write(DataStructure, 12);
```

hasenradball, 2021-05-07:
I never send a struct directly to an Client is there something to consider specially?

bblanchon, 2021-05-07:
`client.write(buffer, sizeof(Messung))` should work as expected.  
I present this option and explain the drawbacks in [this video](https://youtu.be/hAB4TdX8dwM); please, check it out.

closed 2021-05-15 by bblanchon