bblanchon/ArduinoJson #1555: Deserialization of an char

hasenradball, 2021-05-08:
Hi Benoit,

I have an issue during deserealisation of an json.
I get following object ` {"Connect":"2021-05-08T19:30:42.917Z","Relais":"r"}` and I want to extract the `r` as char.
I do it like that:
```
 error = deserializeJson(doc, payload);
        if (error) {
          DBG__PRINT(F(">>> ERROR: Failed to read file, using default configuration!"));
        }
        else {
          if (doc.containsKey("Connect")) {
            auto connection_info = doc["Connect"];
          }
          char relais_request = doc["Relais"][0];
          DBG__PRINT("Relais-Request: ", relais_request);
          //handle_Request(client_id, relais_request);
```

and I get this:
```
 >>> Setup END <<<
 >>> Starte Tor-Öffner...
 [0] Connected from 192.168.178.41 url: /
 [0] get Text: {"Connect":"2021-05-08T19:30:42.917Z","Relais":"r"}
 Relais-Request: ␀
```
I get `NUL` as result

Do you have an idea?

hasenradball, 2021-05-09:
I tried also this 
```
 if (doc.containsKey("Connect")) {
            auto connection_info = doc["Connect"];
          }
          byte relais_request = doc["Relais"].as<char>();
          DBG__PRINT("Relais-Request: ", relais_request);
```
but received 

`0` instead `r`

hasenradball, 2021-05-09:
Ok I had to use an `const char*`

hasenradball, 2021-05-09:
But how to to if I want only an cha type?

bblanchon, 2021-05-09:
Hi @hasenradball,

JSON supports strings and numbers, but not chars.
In fact, I removed support for `char` in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/), because it was ambiguous (signed or unsigned?) and caused errors when constructing strings (see #1498).

In your case, the right way to proceed is to:

1. extract the string
2. make sure the string isn't null
3. extract the character from the string

Like so

```c++
const char* relay_str = doc["Relais"];
char relay_request = relay_str ? relay_str[0] : 0;
```

As an alternative, you can compare the `JsonVariant` with the string you're trying to match:

```c++
if (doc["Relais"] == "r") ...
```

I don't think I can implement a syntax like `char relay_request = doc["Relais"][0];` because:

1. `JsonVariant::operator[](int)` must return a `JsonVariant` (or a proxy)
2. `JsonVariant` is a reference; it must point to a `VariantData` in the memory pool, which we don't have in this case
3. `JsonVariant` doesn't cast to `char` anyway

Best regards,
Benoit

hasenradball, 2021-05-09:
Hi Benoit thanks for your fast answer.

The Background for my Project is that with the values of Relais I go into a switch ... case construction.
Therefore I historically decided to :
use an => r for request. the state of the relais
use an => 1....4 to actuate the different modes of the relais.

In the past I used a Webserver via Http polling, but now I want to switch to a websocket solution.
But I want to keep the Arduinojson usage, ... its great!

best regards
Frank

bblanchon, 2021-05-19:
Hi Frank,

Thank you for the background information about your project.

Do you have any more questions related to this issue, or can I close it?

Best Regards,
Benoit

hasenradball, 2021-05-19:
Hi Benoit,
thank you very much for your support.
actually I have no mor questions.

closed 2021-05-19 by hasenradball