bblanchon/ArduinoJson #1556: JsonDocument is not maintaining char[] data

fscheu, 2021-05-13:
Hi,
I'm facing the same issue described here: https://github.com/bblanchon/ArduinoJson/issues/1191
Assigning a char[] to a JsonDocument doesn't maintain the data.
But in my case it doesn't work either with `operator=()` and adding a cast doesn't fix the error.

Here is my code:

```
   const size_t cap = JSON_OBJECT_SIZE(1);
    StaticJsonDocument<cap> docJson;
    char charArr[20] = "test text";

    Serial.print("test text: ");
    Serial.println(charArr);
    docJson["timestamp"] = charArr;
    //docJson["timestamp"].set((char *)charArr);

    Serial.print("Datos JSON: ");
    serializeJson(docJson,Serial);
```

And the output is

```
test text: test text
Datos JSON: {"timestamp":null}
```

I'm using version v6.17.2.
Thanks for this amazing software

bblanchon, 2021-05-14:
Hi @fscheu,

You must increase the capacity of the `JsonDocument`, so it can store the string.

See:

* [Why is the output incomplete?](https://arduinojson.org/v6/issues/incomplete-output/)

Best regards,
Benoit

fscheu, 2021-05-16:
Hi Benoit!
Thank you for the quick answer. I'm very embarrassed of not found that in the docs.

Thanks again!
Regards,
Fede.-

closed 2021-05-16 by fscheu

bblanchon, 2021-05-16:
You're welcome, Fede!
Don't be embarrassed; the documentation is quite large, and the information you needed was maybe not so easy to find.
Let me know if you have any suggestions to improve the documentation.