bblanchon/ArduinoJson #1557: Float Serialized as Int

aleshark87, 2021-05-13:
With the last update of the library, i get floats variable serialized as ints variable.

This is the code you need:
```
volatile float distance;
volatile int state;

StaticJsonDocument<128> doc;

```

```
doc["type"] = "data";
   doc["state"] = state;
   Serial.println(distance);
   doc["distance"] = distance;
   serializeJson(doc, msg);
   doc.clear();
   Serial.println(msg);
```

From the prints i get:
5.74
{"type":"data","state":2,"distance":5}

Bye

bblanchon, 2021-05-14:
Hi @aleshark87,

Thank you very much for reporting this issue 👍 
I'll fix the bug in branch `6.x` very soon.

As a workaround, you can cast to `float`:

```c++
doc["distance"] = (float)distance;
```

Best regards,
Benoit

closed 2021-05-19 by bblanchon

bblanchon, 2021-07-03:
The fix is included in [ArduinoJson 6.18.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.1)

commit 326da769 (2021-05-18): Added support cv-qualified types in is_floating_point (fixes #1557)
commit 622e7dd2 (2021-05-19): Fixed support for volatile float and double (fixes #1557)