bblanchon/ArduinoJson #1558: deserializejson available field objects

AlirezaAbdiyan, 2021-05-16:
Hi,
In deserializejson how can to find out that there is field objects?
for example:
 {"id":0,"status":1,"mode":1,"delayMs":0,"remainDelayMs":0}}
int id= doc["id"]    ---> id=0
int id1= doc["id1"] ---> id1=0 !!
How can find "id1" is not available?!
Thanks

zeeshanali33096, 2021-05-17:
Hi,

Going through all the docs specified ,
[here](https://arduinojson.org/v6/example/parser/)
[here](https://arduinojson.org/v6/issues/document-contains-null-instead-of-string/)

and still getting null, I am writing this bug.

My Environment - 
OS - Mac Catalina 10.15.7
Arduino IDE - 1.8.13,
Board - ESP32 Wrover
ESP Version - 1.0.6
ArduinoJSON - 6.18.0,

code :

```
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1   //no change
#include <ArduinoJson.h>

String temp = "";
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(3000);
  temp.reserve(200);
}

void loop() {
  // put your main code here, to run repeatedly:
  check_serial();
  delay(50);
}

void check_serial() {
  if (Serial.available()) {
    char x = Serial.read();
    if (x == '\n') {
      Serial.print("input string : ");
      Serial.println(temp);
      check_json(temp);
      temp = "";
    } else  {
      temp += x;
    }
  }
}
void check_json(String &temp){
  StaticJsonDocument<800> doc;  //No help

  DeserializationError error = deserializeJson(doc, temp.c_str());

  // Test if parsing succeeds.
  if (error) {
    Serial.println(F("deserializeJson() failed: "));
   Serial.println(error.f_str());  
    return;
  }

  String xx = "";
  serializeJson(doc, xx);

  String x = doc["ssid_primary"].as<String>();
  const char *p = doc["ssid_primary"];
  Serial.println(xx); // shows correct string
  Serial.println(x);  //returns null
  Serial.println(p);  // blank
  doc.clear();
  

}

```

Serial input -
```"{\"ssid_primary\":\"demo\",\"password_primary\":\"demo1234\"}"```

Where am I going wrong?

bblanchon, 2021-05-19:
Hi @AlirezaAbdiyan,

Please see:

* [JsonDocument::containsKey()](https://arduinojson.org/v6/api/jsondocument/containskey/)

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

AlirezaAbdiyan, 2021-05-20:
Hi @bblanchon 
Thanks

closed 2021-05-20 by AlirezaAbdiyan

bblanchon, 2021-05-21:
You're welcome, @AlirezaAbdiyan!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

AlirezaAbdiyan, 2021-05-21:
Yes, Sure.
Thanks for your helpful library.