bblanchon/ArduinoJson #1559: deserializeJson gives null value

zeeshanali33096, 2021-05-17:
Hi,

Going through all the docs specified ,
[here](https://arduinojson.org/v6/example/parser/)
[here](https://arduinojson.org/v6/issues/document-contains-null-instead-of-string/)

and still getting null, I am writing this bug.

My Environment - 
OS - Mac Catalina 10.15.7
Arduino IDE - 1.8.13,
Board - ESP32 Wrover
ESP Version - 1.0.6
ArduinoJSON - 6.18.0,

code :

```
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1   //no change
#include <ArduinoJson.h>

String temp = "";
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(3000);
  temp.reserve(200);
}

void loop() {
  // put your main code here, to run repeatedly:
  check_serial();
  delay(50);
}

void check_serial() {
  if (Serial.available()) {
    char x = Serial.read();
    if (x == '\n') {
      Serial.print("input string : ");
      Serial.println(temp);
      check_json(temp);
      temp = "";
    } else  {
      temp += x;
    }
  }
}
void check_json(String &temp){
  StaticJsonDocument<800> doc;  //No help

  DeserializationError error = deserializeJson(doc, temp.c_str());

  // Test if parsing succeeds.
  if (error) {
    Serial.println(F("deserializeJson() failed: "));
   Serial.println(error.f_str());  
    return;
  }

  String xx = "";
  serializeJson(doc, xx);

  String x = doc["ssid_primary"].as<String>();
  const char *p = doc["ssid_primary"];
  Serial.println(xx); // shows correct string
  Serial.println(x);  //returns null
  Serial.println(p);  // blank
  doc.clear();
  

}

```

Serial input -
```"{\"ssid_primary\":\"demo\",\"password_primary\":\"demo1234\"}"```

Where am I going wrong?

zeeshanali33096, 2021-05-17:
#1558

bblanchon, 2021-05-17:
Hi @zeeshanali33096,

The bug is not in ArduinoJson but in the input.

The input is:

```json
"{\"ssid_primary\":\"demo\",\"password_primary\":\"demo1234\"}"
```

but it should be:

```json
{"ssid_primary":"demo","password_primary":"demo1234"}
```

As you can see, the object was serialized twice.

You must edit the source code of the sender and remove the extra serialization (for example `JSON.stringify()`, `json.dumps()`, or `json_encode()`, depending on the language you used).

If you cannot change the sender, you'll have to call `deserializeJson()` twice:

```c++
deserializeJson(doc1, temp);
deserializeJson(doc2, doc1.as<const char*>());
```

Duplicates: #932, #1507

Best Regards,
[Benoit](https://donate.benoitblanchon.fr)

zeeshanali33096, 2021-05-17:
Hey @bblanchon !

Thanks man! I figured that!

You can close this.

closed 2021-05-19 by bblanchon