bblanchon/ArduinoJson #1560: IAR compile error (pe070)

Manha96, 2021-05-18:
Hi,

When I include the neweset Version of the ArduinoJson Library (v6.18.0)  into my IAR workbench (v8.50.9), I always get the same error:

**Error Pe070:**
_Error[Pe070]: incomplete type is not allowed. Detected during instantiation of "inline bool ArduinoJson6180_11::VariantRef::set(T const &) const [with T=signed char]" at line 94_ 

**Line 94 in the Library:**
`template <typename T>
  FORCE_INLINE bool set(const T &value) const {
    return Converter<T>::toJson(value, *this);
  }`

After trying several things such like creating new iar project, try c instead of cpp etc., I finaly checked out the previous version (v6.17.3). With this version I didn't get any error.

- Is this iar compile error already known? 

- If yes, are the current branches tackle this problem?

I hope my description was clear enough. I didn't wrote any code with the library. I have just included the header file.
Suprisingly, I don't get any error with visual studio neither. 

BR

bblanchon, 2021-05-18:
Hi @Manha96,

Thank you very much for reporting this issue.
I'm pretty sure that I wrote legal C++ code here; this is probably a limitation of the IAR compiler.
I'll try to reproduce the issue here and, hopefully, find a workaround.

Best regards,
Benoit

bblanchon, 2021-05-18:
I managed to reproduce with a trial license of IAR Embedded Workbench for ARM 9.10.1.
Removing [the faulty method](https://github.com/bblanchon/ArduinoJson/blob/36bcc7b355b5acb5b7b85fd952fe29c6ec377522/src/ArduinoJson/Variant/VariantRef.hpp#L91-L95) fixes the build for me.
Can you confirm that it works for you as well?

Manha96, 2021-05-18:
Hi Benoit

Thank you for your immediate reply. Yes, it seems that if I remove this methode the error disapears. 
Is there any disadvantage if I remove the code?

And perhaps just for personal interest, do you probably know the source of this problem? 

BR,
Manuel

bblanchon, 2021-05-18:
Hi Manuel,

As you can see, this method is a shim that displays a deprecation warning; you only need it if you use the deprecated behavior, i.e., if you assign a naked `char` to a `JsonVariant`.

For the record: the compiler complained that the type `Converter<signed char>` was declared (in [Converter.hpp](https://github.com/bblanchon/ArduinoJson/blob/36bcc7b355b5acb5b7b85fd952fe29c6ec377522/src/ArduinoJson/Variant/Converter.hpp#L9)) but not defined; hence, it was deemed "incomplete" (the definition comes later in [ConverterImpl.hpp](https://github.com/bblanchon/ArduinoJson/blob/36bcc7b355b5acb5b7b85fd952fe29c6ec377522/src/ArduinoJson/Variant/ConverterImpl.hpp#L38)).
I guess other compilers were happy with that because they instantiated the template lazily, whereas IAR seems to do it earlier.

Anyway, thanks for the confirmation; I'll publish a patch very soon.

Best regards,
Benoit

closed 2021-05-19 by bblanchon

bblanchon, 2021-07-03:
The fix is included in [ArduinoJson 6.18.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.18.1)

commit af3bb131 (2021-05-19): Fixed error Pe070 "incomplete type is not allowed" on IAR (fixes #1560)