bblanchon/ArduinoJson #1563: Linker error when using DynamicJsonDocument()

Manha96, 2021-05-20:
Hi,

Sorry for reporting another issue. I serialized and deserialized a StaticJsonDocument with all available Datatypes and it absolutely worked well. Since I tried to switch to dynamic allocation to keep the stack size small (big Json Documents), I always get the same error message from the Linker:

**Code:**
std::uint8_t size = 10;
DynamicJsonDocument doc(size);

**Linker error:**
_Error[Li005]: no definition for "__basic_free" [referenced from heaptramp0.o(dl7M_tln.a)]_ 

Do you probably know whre __basic_free is used and where we have to define it?
The error appears in all v6.x versions and I am still using Iar workbench V8.50.9.

Thank you very much,
Manuel

bblanchon, 2021-05-21:
Hi Manuel,

I'm sorry, but I cannot help you this time since it's not a problem related to ArduinoJson.

`DynamicJsonDocument` uses the standard `malloc()` and `free()` functions; I don't know why the linker cannot find them, but I suppose there is a problem in your project's configuration.
You can probably reproduce the error with the following program:

```c++
int main() {
  free(0);
}
```

After moving ArduinoJson out of the way, you should send a question to IAR support or Stack Overflow.

Best regards,
Benoit

Manha96, 2021-05-21:
Hi Benoit

I just tried that and I totally agree. It must be an error on our side.
I will have a look on the project configuration.

Best regards,
Manuel

closed 2021-05-28 by bblanchon