bblanchon/ArduinoJson #1564: Problem using StaticJsonDocument template size as a template argument in a function

Kuenlun, 2021-05-20:
Hi,
I'm trying to make a google calendar API.
I've wrapped the `deserializeJson` function inside a function called parse_json to simplify error handling for my API.

At first I used the function with hardcoded size for the StaticJsonDocument and no problem at all:
_Code header:_
`ApiError parse_json(StaticJsonDocument<COMPILER_TIME_KNOWN_VALUE>& json_doc, String& response);`

But when I use a template argument for the size I get a compiler error:
_Code:_
`template<std::size_t SIZE>`
`ApiError parse_json(StaticJsonDocument<SIZE>& json_doc, String& response);`

_Compiler error:_
`(.literal._Z5setupv+0x14): undefined reference to ApiError parse_json<1024u>(ArduinoJson6180_91::StaticJsonDocument<1024u>&, String&)`

BTW it only happens when the function is inside a **cpp file** different to the arduino main file. If I put the function on the .ino file it works no problem.

So ,how can I **pass the StaticJsonDocument size value as a template argument**?
I've seen #977 but don't know if is exists a similar solution or not.

Destination platform is: "DOIT ESP32 DEVKIT V1"
Arduino IDE version is: "1.8.14"

bblanchon, 2021-05-20:
Hi @Kuenlun,

You should be able to use a `JsonDocument` parameter:

```c++
ApiError parse_json(JsonDocument& json_doc, String& response);
```

This is much better than the function template because the same code works for all sizes and also supports `DynamicJsonDocument`.

Best regards,
Benoit

Kuenlun, 2021-05-20:
Thanks a lot, it works perfectly. And also the function declaration has get smaller and clearer.
Thanks for all the work you've put in this project.

closed 2021-05-20 by Kuenlun